google/aiyprojects-raspbian #426: AIY-Vision for Ubuntu Mate

jessecha, 2018-07-04:
Is there anyway I can make the vision_devcomm file appear on a fresh install of an Ubuntu Mate OS?
Or more bluntly speaking, how can I get the vision bonnet to communicate with the RPi with a Ubuntu environment?
I want to use Ubuntu instead of Raspbian because its easier to install packages for ROS.

dmitriykovalev, 2018-07-16:
We are going to update installation instructions (HACKING.md file) in the next SD card release. There is a low-level Linux driver called `aiy-vision` to communicate with the bonnet.

jessecha, 2018-07-17:
Thank you. Raspbian is such a painful OS to use because a lot of packages have to be compiled from source.

manoj7410, 2018-08-10:
@jessecha HACKING.md file has been updated. You can get it from : https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

jessecha, 2018-08-10:
The update does not enable me to use ubuntu mate.

dmitriykovalev, 2018-08-10:
We haven't tried anything except Raspbian for now. It would be great if you can help us to figure out what needs to be fixed for Ubuntu Mate.

jessecha, 2018-08-11:
The issue seems to be with the rpi 3B board not being able to find the spi_comm file in the vision bonnet.

dmitriykovalev, 2018-08-13:
`spi_comm` file comes from `aiy-vision` driver.  Did you manage to install it?

closed 2020-11-20 by dmitriykovalev