google/aiyprojects-raspbian #429: Setup Issue with new install of Voice Assitant

briankmcm, 2018-07-07:
I recently acquired Voice Kit. I realized it was version 1.0. Went through the set up got through to Chapter 5 step 13 when I ran into an issue. When I enter as instructed "src/assistant_library_demo.py" in the development terminal it returns the following;
"bash: src/assistant_library_demo.py: No such file or directory" 
I went to the folder where that code is located and tried running it from there and got an ImportError.
Need help.
Thank you Brian

dmitriykovalev, 2018-07-10:
@briankmcm Did you try to follow instructions at https://aiyprojects.withgoogle.com/voice-v1/ ?

briankmcm, 2018-07-11:
I followed the instruction booklet that came with the kit.
Thanks,

manoj7410, 2018-07-11:
@briankmcm 

Download the latest sdcard image from https://aiyprojects.withgoogle.com/help/#updating-your-sd-card and install it on your sdcard.

You can follow the instructions given at : https://aiyprojects.withgoogle.com/voice/#connect (These will work for Voice V1 as well) to make your assistant work.

briankmcm, 2018-07-11:
Will give this a try.
Thank you!!

closed 2018-07-28 by dmitriykovalev