google/aiyprojects-raspbian #430: No module named aiy.assistant.grpc   No module named aiy.audio

tiancaipipi110, 2018-07-07:
Google AIY voice installation went fine, testing sound passed. However when I try to run `python assisstant_grpc_demo.py`  it shows error   `No module named aiy.assistant.grpc`
when run `python cloud_speech_demo.py`   it shows error  `No module named aiy.audio`
Any idea why?

closed 2018-07-10 by tiancaipipi110

slloyd88, 2018-09-13:
I get the same thing if I try to run the tests as root instead of pi

manoj7410, 2018-09-13:
@slloyd88 That is because environment of root user is not setup as per the requirement. You can see the difference between environment of "root" and "pi" by running the command printenv (Once with root and then with pi).

slloyd88, 2018-09-13:
Is there a way for me to setup root also?  I assume you mean that root needs all the INFINALITY_ environment keys set.

dmitriykovalev, 2018-09-13:
@slloyd88 Could you please point to the instructions which require root usage?