google/aiyprojects-raspbian #431: Spicomm error testing compiled binary model (both default model and own model)

Aileenlingyu, 2018-07-11:
Hi, I have successfully compiled my own model into binaryproto, I tried using test_run_model_on_bonnet.py to test it but received the following error:
[image]

I followed the instruction [here](https://github.com/google/aiyprojects-raspbian/issues/335) in issue #335 and updated my SD card image to the latest one, but the error is still there.

I also tried to test the built-in default graph in /opt/aiy/models/face_detection.binaryproto, without making any changes, and got the same error. Also if I test on the squeezenet_160res_5x5_0.75.binaryproto, same error message will show after printing out some inference result like below:
[image]

Could anyone please advise? Many thanks! 

By the way, I can run all the built-in demos without no error messages.

ritwikbiswas, 2018-07-14:
What does your full call to `test_run_model_on_bonnet.py` look like?

Aileenlingyu, 2018-07-14:
Hi @ritwikbiswas, I am using the following command to run:
python3 ~/AIY-projects-python/src/examples/vision/test_run_model_on_bonnet.py \
  --model_path ~/models/my-graph.binaryproto \
  --input_height 64 \
  --input_width 64

ritwikbiswas, 2018-07-14:
What about your labels path? And your input and output layers?

Does you input heigh and width match that of your input heigh/width when you compiled the model?

weiranzhao, 2018-07-16:
@Aileenlingyu 
It seems like you are building a model with 64x64 input, I don't think this is a supported configuration for Vision Kit.
https://aiyprojects.withgoogle.com/vision/#makers-guide--tensorflow-model-compiler

Aileenlingyu, 2018-07-16:
Hi @ritwikbiswas, I am using a regression model rather than the classification, so I do not have labels. Yes the input height and width are exactly same as the values for compiling. Is it possible for me to use a regression model? Thanks much!

Aileenlingyu, 2018-07-17:
Hi @weiranzhao , I found the following instruction [here](https://aiyprojects.withgoogle.com/vision/#constraints) suggesting that as long as the input size is a multiple of 8, it will be fine, 

> Model takes square RGB image and input image size must be a multiple of 8.
> Note: Vision Bonnet handles down-scaling, therefore, when doing inference, you can upload image that is larger than model's input image size. And inference image's size does not need to be a multiple of 8.

is that correct or not? Many thanks!

weiranzhao, 2018-07-17:
@Aileenlingyu That is like a general guideline. It also depends on the model structure you used and whether operator configuration is supported. The reason I'm concerned about using a 64x64 model input is that if your network is deep, there might be intermediate tensors that have size 2x2xdepth, and  if you use a 3x3 conv kernel on that input, Vision Kit can not handle that case.

Aileenlingyu, 2018-07-17:
Hi @weiranzhao, I am using only 3 convolution layers as below, 
[image]

I was wondering that, is there any other possible reason I may consider for my case, i.e., there is no error when compiling frozen pb into binaryproto, but getting spicomm error and camera no data error when testing the model?

Thanks a lot!

weiranzhao, 2018-07-17:
@Aileenlingyu Could you try to get rid of LRN during inference and try again? I don't think LRN is a supported operator.
https://aiyprojects.withgoogle.com/vision/#makers-guide--tensorflow-model-compiler

Aileenlingyu, 2018-07-17:
Thanks much, will try!

Aileenlingyu, 2018-07-20:
@weiranzhao , thank you for your advice! I removed the LRN layer and tried with a simple sub-network, and I am able to successfully pass the test script now.

closed 2018-07-20 by Aileenlingyu