google/aiyprojects-raspbian #432: PiCameraRuntimeError: No data recevied from sensor

barnwal-paras, 2018-07-12:
Tried running mobilenet based classifier on vision kit. I was getting following error. Although demo script is running perfectly.

`
pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ ~/AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py   --model_path ~/retrained_graph.binaryproto   --label_path ~/retrained_labels.txt   --input_height 160   --input_width 160   --input_layer input   --output_layer final_result   --preview
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 14] Bad address
 
During handling of the above exception, another exception occurred:
 
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 110, in main
    for i, result in enumerate(camera_inference.run()):
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 76, in run
    yield self._engine.camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 257, in camera_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 152, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError
 
During handling of the above exception, another exception occurred:
 
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 129, in main
    'ascii','backslashreplace').decode('ascii')
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 87, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 79, in close
    self._engine.stop_camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 263, in stop_camera_inference
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.
 
During handling of the above exception, another exception occurred:
 
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 135, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/vision/mobilenet_based_classifier.py", line 132, in main
    camera.stop_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
 `
I believe the problem is not with connections as all the demo script are working without any problem.
Thanks in advance for your help in troubleshooting.

Aileenlingyu, 2018-07-13:
Same problem here.

barnwal-paras, 2018-07-13:
BTW which image are you using? It might be the case that this is specific to latest version of the image.

dmitriykovalev, 2018-07-14:
@barnwal-paras which demo script is working for you? If that's a JoyDemo, did you disable it before running other demos?

ritwikbiswas, 2018-07-14:
I was running into this same exact issue until I updated to the latest version of the image. From your vision kit run:

`sudo apt-get update; sudo apt-get upgrade; sudo reboot`

dmitriykovalev, 2018-07-15:
@barnwal-paras Did you run the same update command as @ritwikbiswas  did?  Updating like this could cause a problem, please try to use our SD-card image if possible.

JakeBMalis, 2018-07-15:
I am having the same issues as the folks above.

I am just starting out on Tensorflow & AIY Vision Kit. I am able to completely build Tensorflow for Poets on Colab as well as on a GCE instance (NVIDIA K80 GPU). I do this with 160 size and 0.5.

I then compile the PB file into a BinaryProto and upload it (and the retrained_labels.txt) to the RPI. When I run the mobilenet_based_classifier.py file, the program crashes after 5 seconds of seeing what the camera sees and I get the following 3 errors:
PI Camera Runtime: No Data received from sensors
AIY Vision Inference: Camera is not running
AIY Drivers Spiccom internal error

Any ideas?

Thanks.

weiranzhao, 2018-07-16:
@JakeBMalis 
Did you update your SD card to the latest version? To support deploying retrained model, you need to use the latest version of SD card.
https://aiyprojects.withgoogle.com/help/#updating-your-sd-card

Could you paste the output of 
cat /etc/aiyprojects.info

and dpkg -l aiy*?

Thanks

Aileenlingyu, 2018-07-16:
Hi @dmitriykovalev , I update my SD card image as as @ritwikbiswas did, but I am still having the same problem, did you mean we should reflash SD card directly?
[image]

Thanks much!

dmitriykovalev, 2018-07-17:
@Aileenlingyu Right, reflash SD card directly from the latest image file from https://github.com/google/aiyprojects-raspbian/releases

Aileenlingyu, 2018-07-17:
Hi @dmitriykovalev , thank you for the suggestion! After re-flashing my SD card, I am now able to successfully test almost all built-in models such as mobilenet_v1_160res_0.5_imagenet.binaryproto, while I am still having the error when I tried the following cmd:
`python3 test_run_model_on_bonnet.py --model_path face_detection.binaryproto --input_height 1640 --input_width 1232`

Could you please advise how should I test the built-in face detection model using the correct command? What kind of input tensor size shall I use here?

Thank you!

weiranzhao, 2018-07-17:
@Aileenlingyu 
face_detection.binaryproto is a very special case, i.e., it is a collection of models. That is why it does not have a corresponding TF frozen graph. But if you really want to test that, you can try to change name='test_run_model', to  name='FaceDetection' in test_run_model_on_bonnet.py. You can use the height and width you were using.

Aileenlingyu, 2018-07-17:
Hi @weiranzhao , thanks a lot! I am now able to test face detection model as well.

closed 2018-07-28 by dmitriykovalev