google/aiyprojects-raspbian #433: Load multiple models for Inference?

Aileenlingyu, 2018-07-13:
Hi, I was wondering that, is that possible for me to load multiple models together for camera inference? I need to feed the output of one model as an input to another model while the camera keeps streaming, but I got error like 'Cannot load model during processing' 'Cannot unload model during processing', is there any suggestion to solve it? Very grateful!

dmitriykovalev, 2018-07-14:
@Aileenlingyu That's kind of possible, depends on your use case. Please look into `InferenceEngine` class inside `inference.py`. You can load multiple models by calling `load_model()` several times. Existing `CameraInference` and `ImageInference` are just wrappers around `InferenceEngine`.

ritwikbiswas, 2018-07-14:
@dmitriykovalev 
> That's kind of possible, depends on your use case.

What use cases will this work for? I'm trying to achieve a similar thing but feeding a camera stream into two different models in the same script. Looking at the source code (`inference.py`) it looks like calling load_model() multiple times will overwrite the previous loaded model.

Is there a way to run two models simultaneously?

dmitriykovalev, 2018-07-15:
@ritwikbiswas Each model has unique identifier (`model_name` in python code), it's possible to load multiple models at the same time and then use that identifier to reference them. The tricky part is that you can only use one model for camera inference (at least for now), you cannot process each camera frame with more than one model.

Aileenlingyu, 2018-07-15:
Hi @dmitriykovalev , thanks for the suggestion! Is that possible to process the each camera frame with only one model but add a post-processing using another model?

dmitriykovalev, 2018-07-16:
@Aileenlingyu Currently this is not possible but could be implemented in future. I'll mark this issue with "enhancement" label.

llevai314159, 2018-10-27:
Any update on this?

dmitriykovalev, 2018-10-28:
This is something that can be done but that's not the highest priority right now.