google/aiyprojects-raspbian #434: longer short flex

wuhland, 2018-07-16:
I would like to use the AIY vision setup in conjunction with another peripheral attached in between the Vision Bonnet and the RPi. Sadly the short flex is a wee bit short to accommodate this. Are there somewhat longer versions of this flex available anywhere?

dmitriykovalev, 2018-07-24:
Any of these 22POS 0.5MM pitch FFC cables should work:

https://www.digikey.com/products/en/cable-assemblies/flat-flex-ribbon-jumper-cables/457?k=FFC&k=&pkeyword=FFC&pv1989=0&FV=1640055%2Cii22%7C2166%2Cffe001c9%2C40800d1&quantity=0&ColumnSort=0&page=1&pageSize=25

andrewtvuong, 2018-07-26:
@dmitriykovalev Hi I am trying to use the Pi 3 B with the vision bonnet. Can you recommend a flex cable between the Pi 3 B and bonnet that has a short enough connector that it would fit? The USB ports are too close to the bonnet for the standard cables.

I even clipped the peeled hard black piece in half but it still seems too squished, may have shorted the board because it wouldn't even power on.

Thanks for any suggestions!

closed 2018-07-28 by dmitriykovalev