google/aiyprojects-raspbian #437: AIY Vision Accuracy

JakeBMalis, 2018-07-22:
I am using Colaboratory and starting with "tensorflow-for-poets-2", but using Playing Cards instead for a proof-of-concept. After training (re-training) 60 King of Spades and 60 Queen of Spades, testing a single card of either shows 100% accuracy when running the "label_image" scripts in colab.

When compiling and moving to the AIY Vision Kit, it only is accurate when the card is held inches from the camera (around 90%). If I move the card back 2-3 feet (so that the card takes up 10% to 20% of the monitor), the accuracy drops down to zero. It thinks both cards are the King of Spades. We want it to work with cards not having to be close to the camera.

Is the issue that we only used 50 of each card and should be using 1000? Or could it be something completely different? We also tried moving from "MobileNet_v1_0.50_160" to "MobileNet_v1_1.0_192" and that didn't make a difference.

Also, is "mobilenet_based_classifier.py" the way to go? Would "object_detection.py" work better? This part is a bit confusing to me.

Any ideas would be appreciated.

Thank you.

weiranzhao, 2018-07-22:
@JakeBMalis 

When you trained your network, what kinds of images are you using? Are the 60 King/Queen of Spades taken 2-3 feet away from the camera? I'm assuming 100% is the accuracy on training set, what is the accuracy on your testing data set?

Also, when the card is only taking 10% to 20% of the scene, background plays an important role as well, i.e., if you train 60 King/Queen of spaces taken 2-3 feet away from the camera, but with a pure color background. But when you test it, you give it a more colorful background, the accuracy could drop a lot as well.

In your case, it's best to stick with mobilenet_based_classifier.py.

JakeBMalis, 2018-07-27:
@weiranzhao 

Hi. I'd love to ask you a few more questions.

- So, for my proof-of-concept, the playing cards will always be in a distance with various backgrounds, so should the training pictures all be of the playing cards at a distance with various backgrounds?
- What is the best camera resolution for training?
- How many cards of each do you think I'll need?
- If I need to detect multiple playing cards at the same time, will "MobileNet_v1_1.0_192" work for me?
- Is there a way for me to create a red box around the playing card like I've seen in some examples? Do you happen to have code for that?
- What's the difference between "MobileNetV1 + SSD" and MobileNet_v1?
- What is the real difference between "mobilenet_based_classifier.py" and "object_detection.py"?

Thank you.

weiranzhao, 2018-07-30:
@JakeBMalis 

So, for my proof-of-concept, the playing cards will always be in a distance with various backgrounds, so should the training pictures all be of the playing cards at a distance with various backgrounds?
Yes

What is the best camera resolution for training?
The model itself takes 160x160 input, or 192x192 depends on your setting. So the default resolution is definitely enough. 

How many cards of each do you think I'll need?
Depends on what type of accuracy you want to achieve eventually. But I'd say at least 80 ~ 100 each.

If I need to detect multiple playing cards at the same time, will "MobileNet_v1_1.0_192" work for me?
Not really

Is there a way for me to create a red box around the playing card like I've seen in some examples? Do you happen to have code for that?
You will need a detection model in this case, it is a bit harder to train and more labor to clean and collect the data. You can follow the instructions here once you have the data for training. But I'd suggest to start with a classification model.
"How to train and deploy a customized object detection model trained with TF’s object detection API?" at https://aiyprojects.withgoogle.com/vision/#makers-guide--tensorflow-model-compiler

What's the difference between "MobileNetV1 + SSD" and MobileNet_v1?
SSD models are for object detection, which will put a bounding box around objects it detects

What is the real difference between "mobilenet_based_classifier.py" and "object_detection.py"?
mobilenet_based_classifier.py is a general script that runs mobilenet based classification model
object_detection.py is a dedicated script for running our person, cat dog detector.

Is there a way to label the objects that we want to train the network with?
There are image labelling service online, you can try to search that.

closed 2018-11-17 by dmitriykovalev