google/aiyprojects-raspbian #438: Crash in the bonnet_model_compiler.par on the mobilenet_v1_1.0_128_quant_frozen

dchichkov, 2018-07-24:
One of your AIY users here, need to be able to run exact same model on AIY and x86/GPU.  What kind of quantization does it do?  Or can I feed already quantized model into bonnet_model_compiler.par?   The thing crashes on me, if I'm trying that:

```
wget http://download.tensorflow.org/models/mobilenet_v1_2018_02_22/mobilenet_v1_1.0_128_quant.tgz
 tar -xf mobilenet_v1_1.0_128_quant.tgz
./bonnet_model_compiler.par --frozen_graph_path=mobilenet_v1_1.0_128_quant_frozen.pb   --output_graph_path=mobilenet_v1_1.0_128_quant_frozen.binaryproto --input_tensor_name="input" --output_tensor_names="MobilenetV1/Predictions/Reshape_1" --input_tensor_size 128 --debug
2018-07-23 18:19:17.978971: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 598 operators, 889 arrays (0 quantized)
2018-07-23 18:19:17.999724: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 598 operators, 889 arrays (0 quantized)
2018-07-23 18:19:18.479095: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 86 operators, 144 arrays (0 quantized)
2018-07-23 18:19:18.480780: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 86 operators, 144 arrays (0 quantized)
2018-07-23 18:19:18.482054: I external/org_tensorflow/tensorflow/contrib/lite/toco/toco_tooling.cc:324] Estimated count of arithmetic ops: 0.374532 billion (note that a multiply-add is counted as 2 ops).
2018-07-23 18:19:18.532240: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 142 operators, 200 arrays (0 quantized)
2018-07-23 18:19:18.535039: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 142 operators, 200 arrays (0 quantized)
2018-07-23 18:19:18.549292: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 86 operators, 144 arrays (0 quantized)
2018-07-23 18:19:18.550929: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 86 operators, 144 arrays (0 quantized)
2018-07-23 18:19:18.552198: I external/org_tensorflow/tensorflow/contrib/lite/toco/toco_tooling.cc:324] Estimated count of arithmetic ops: 0.374532 billion (note that a multiply-add is counted as 2 ops).
2018-07-23 18:19:18.580399: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 142 operators, 200 arrays (0 quantized)
2018-07-23 18:19:18.583215: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 142 operators, 200 arrays (0 quantized)
2018-07-23 18:19:18.597478: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 85 operators, 142 arrays (0 quantized)
2018-07-23 18:19:18.599067: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 85 operators, 142 arrays (0 quantized)
2018-07-23 18:19:18.600300: I external/org_tensorflow/tensorflow/contrib/lite/toco/toco_tooling.cc:324] Estimated count of arithmetic ops: 0.374532 billion (note that a multiply-add is counted as 2 ops).
2018-07-23 18:19:18.637575: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before Removing unused ops: 142 operators, 200 arrays (0 quantized)
2018-07-23 18:19:18.640360: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before general graph transformations: 142 operators, 200 arrays (0 quantized)
2018-07-23 18:19:18.654593: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] After general graph transformations pass 1: 85 operators, 142 arrays (0 quantized)
2018-07-23 18:19:18.656206: I external/org_tensorflow/tensorflow/contrib/lite/toco/graph_transformations/graph_transformations.cc:39] Before dequantization graph transformations: 85 operators, 142 arrays (0 quantized)
2018-07-23 18:19:18.657443: I external/org_tensorflow/tensorflow/contrib/lite/toco/toco_tooling.cc:324] Estimated count of arithmetic ops: 0.374532 billion (note that a multiply-add is counted as 2 ops).
2018-07-23 18:19:18.664533: F export_mognet.cc:572] Can't export FakeQuant nodes to Mognet format; consider passing --drop_fake_quant. In tfmini_test, consider passing tensorflow_only = True.

```

weiranzhao, 2018-07-25:
@dchichkov 
AIY vision kit only supports float inference engine. You can try the non-quantized version of mobilenet.
https://github.com/tensorflow/models/blob/master/research/slim/nets/mobilenet_v1.md

dchichkov, 2018-07-26:
When the model is running on the device, is the original FP32 model being executed, or some quantized version?   I need to be able to run exact same model on AIY and x86/GPU for verification purposes.

weiranzhao, 2018-07-26:
@dchichkov 
Under the hood, it uses Fp16 to run inference. But there will always be numeric difference because of implementation details, e.g. use of single float or double as accumulator, use of hardware convolution block, and etc. 

Depending on what you want to achieve, if you want to verify the statistical behavior stays the same when deploying on Vision Kit, I'd recommend to check precision/recall on a small data set.

closed 2018-10-05 by dmitriykovalev