google/aiyprojects-raspbian #439: compiler doesn't support tf.keras.layers.Flatten

algila, 2018-07-24:
Any plan to update the compiler managing also `tf.keras.layers.Flatten` ?
That function contain the function np,Prod that is not managed by the compiler `bonnet_model_compiler_2018_04_26`

algila, 2018-07-25:
I replaced `Flatten` with the instruction `Reshape` but the compiler provide an other error message

`2018-07-25 18:57:52.697603: F external/org_tensorflow/tensorflow/contrib/lite/toco/export_tensorflow.cc:1863] Unhandled operator type TensorFlowTile
`

algila, 2018-08-01:
I replaced `tf.tile` with the instruction `tf.expand_dims` as below 

```
#prior_boxes_tensor = tf.tile(prior_boxes_tensor, pattern)  
prior_boxes_tensor = tf.expand_dims(tf.ones([tf.shape(x)[0], 1]), 1) * prior_boxes_tensor 
```
but  the compiler (bonnet_model_compiler_2018_04_26) provide an other error message. I really hope that this reporting will give the message that the compiler should be updated to simplify the life or the final user. Waiting that any idea how to fix it ?

`2018-08-01 20:33:09.622163: F external/org_tensorflow/tensorflow/contrib/lite/toco/export_tensorflow.cc:1863] Unhandled operator type TensorFlowMax`

bowu-google, 2018-08-03:
May I ask what's the model architecture?
Is this one of the supported architectures listed on Vision Kit website:
https://aiyprojects.withgoogle.com/vision/
(search for "TENSORFLOW MODEL COMPILER")

algila, 2018-08-03:
Hi,
No I am using an ssd 256x256 based on mobilenet alpha 0.125 written in keras and converted in TensorFlow. The graph is too big so I am not able to find where the TF.Max is used.

bowu-google, 2018-08-06:
hmm, 

i see. you trained with Keras and ten converted to TF.
we only tested the workflow on models trained with TF.
there is no guarantee that the model trained with other framework, then converted to TF, could pass the compiler. 
could you please train the model with say TF/slim?

closed 2018-11-17 by dmitriykovalev