google/aiyprojects-raspbian #440: Issues with Vision Kit & demos

davejavu1969, 2018-07-26:
Been having several issues running the Vision Kit and associated demos. I am having repeated problems with the unit freezing / crashing / failing to boot. Will list them individually - all help gratefully received. 

1. Boot screen - what should actually be displayed on boot? (not sure why this would change) I have seen four different boot screens. (i) Raspberry logo & flashing cursor (ii) Standard terminal boot text (iii) Raspbian splash boot screen (iv) Grey screen with three little animated dots. Which one I see is apparently random. 

2. I am using 'official' Pi charger. Both Pi and Vision board run *extremely* hot, so hot in fact I suspect this was causing the repeated crashes I am experiencing so have removed from outer box and am continuing testing. Have others experienced this? All cables are correctly connected (I can see camera preview in demos) I have also already done the overvolt = 4 tweak as suggested in another post. 

3. Joy demo has never worked. I sometimes got three beeps on boot, sometimes not. Very random and annoying again. Even when the unit fully boots correctly and the beeps sound, the joy demo doesn't appear to do anything. The LED doesn't light up etc. 

4. Image Classification Camera Demo. Once running, should the 'detected object list' keep streaming? – it doesn’t, I get about 20 lines and then it crashes out. Additionally, Ctrl-C doesn’t actually stop process or close preview, so can’t then re-run as I get 'no resources' error msg. Ctrl-Z stops process, but leaves preview window open. Have not identified process in top to kill it. 

5. Face Detection Camera Demo. Preview window runs, but no text displayed in terminal window. Same issues with Ctrl-C when trying to close process. 

Many thanks.

DP.

manoj7410, 2018-07-27:
@davejavu1969 Try to edit /boot/config.txt and uncomment #arm_freq=800.

This can also help in preventing this crash.

andrewtvuong, 2018-08-01:
If you're using a not so good SD card, that could be an issue.

Have you tried a fresh image? AIY 4-13-18? (don't update and upgrade)

If not of the above works; can you try to make sure everything is connected properly?

2. Can you try a regular usb cable with a cellphone charging block and see the temp?

4. I get about 20 seconds before crashing, but my CTRL+C stops it fine. Can you list your processes from Top? just a screenshot or paste.  try pkill python3 to stop.

rangodjango, 2018-09-25:
Hi there, I am newbie in raspi and currently using vision kit v1.1. I did add some new lines in joy_detector_demo.py so that it able to connect with GPIO and test it out. But when I run command ./joy_detector_demo.py in terminal, the button leds looks like mixed flip-flop when pointing the camera to the face...

There are also errors that I got as below:
mmal:mmal_vc_port_enable: failed to enable port vc.null.sink:in:0(OPQV):ENOSPC
mmal:mmal_port_enable:failed to enable connected port(vc.null_sink:in:0:(OPQV)0x1eba440 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
picamera.exc.PiCameraMMALError:Failed to enable connection:Out of resources

How can I resolve this? 

Thanks

manoj7410, 2018-09-25:
@rangodjango Seems like there is already a process running which is using camera module. Can you make sure that the joy_detection_demo has been stopped which starts at every boot.

Run the command : "sudo systemctl stop joy_detection_demo" and then start your joy demo. At least this ENOSPC should go away.

rangodjango, 2018-09-25:
Your're right. It should turn off the joy_detection_ demo first. Just realized that we run this code first before run others demo as well. Now it's working. No more ENOSPC. Thanks @manoj7410

dmitriykovalev, 2018-10-01:
@rangodjango Looks like something happened with your setup. You can flash SD card again and start from the working state.

closed 2020-05-13 by dmitriykovalev