google/aiyprojects-raspbian #441: Any diagnostics to test the health of the Vision Bonnet?

avyii, 2018-07-26:
With all the tinkering and the bonnet being inherently passively operating, I fear I've slightly damaged the bonnet without really knowing. Are there any software diagnostics I could run to check its health?

manoj7410, 2018-07-27:
@avyii Does any of your demo work now ?
What happens when you run leds_example and buzzer_demo.py ?

Do you see "Myriad Ready" in your dmesg output ?

avyii, 2018-07-27:
yes they still work though it seems a bit slower; I was wondering if I can monitor its wear. Just realized it's a pretty expensive chip to mess around with that also isn't so obvious in its functions.

dmitriykovalev, 2018-07-28:
There are no special diagnostic tools, please let us know if you are facing any specific errors, e.g. inference examples do not work. Raspberry Pi itself can have issues as well.

closed 2018-08-16 by dmitriykovalev