google/aiyprojects-raspbian #442: Not Seeing Bounding Boxes on face_detection_camera.py and Crashing after short while w/ or w/o ffmpeg streaming

galwayhooker, 2018-07-28:
For some reason; I am unable to see the bounding boxes in face_detection_camera.py while I stream the video. I have not seen any bounding boxes on anything yet. 

I do get inference std output in form of text of face detection.

I have modified the face_detection_camera.py file with start_recording() instead of start_preview(). I choose to output to ffmpeg by piping. Video streaming is successful but without bounding boxes and eventually fails from what I believe mismatching between output, piping, and streaming when annotation tries to enter the mix. 

Sorry I'm not sure what to make out of it or have any solid clue what is causing failure; which I believe is related to bounding boxes annotation. The code works fine when annotation is removed.

```
ffmpeg = subprocess.Popen([
        'ffmpeg', (input flags), 'rtmp://xxxxxxxxxxxx',
        ], stdin=subprocess.PIPE)

with PiCamera() as camera:
        camera.sensor_mode = 4
        camera.resolution = (1640, 1232)

        # Start the camera stream.
        camera.framerate = 25
        camera.start_recording(ffmpeg.stdin, format='h264', bitrate=5000000)
```

```
Traceback (most recent call last): 54808kB time=00:01:29.60 bitrate=5010.7kbits/s speed=0.858x    
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 137, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, buf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 82, in main
    for i, result in enumerate(inference.run()):
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 76, in run
    yield self._engine.camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 257, in camera_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 152, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 90, in main
    print('Iteration #%d: num_faces=%d' % (i, len(faces)))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 87, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 79, in close
    self._engine.stop_camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 263, in stop_camera_inference
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 189, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 96, in <module>
    main()
  File "face_detection_camera.py", line 92, in main
    camera.stop_recording()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
```

dmitriykovalev, 2018-07-30:
Bounding boxes are only displayed on HDMI output. If you are streaming video then you have to transfer bounding box information yourself and display them on top of your video stream. 

Regarding your exception traces, have you disabled JoyDemo service before running example code?

galwayhooker, 2018-07-31:
Oh no wonder, didn't expect it to be HDMI output only. Is there a centralized documentation on the functions? I am curious about the implementation. 

Yes I have disabled the service completely.

dmitriykovalev, 2018-07-31:
Just check the python code, everything is right there: `face_detection.py` has functions to get bounding boxes from inference result. Are you still having that exception? Can you share the whole piece of code in this case?

closed 2018-08-16 by dmitriykovalev