google/aiyprojects-raspbian #443: Installation on fresh stretch Raspbian image not working

patoberli, 2018-07-29:
Hi All
I've got the same issue like the closed issue https://github.com/google/aiyprojects-raspbian/issues/309. 
It seems the Hacking.md instructions are completely outdated, as there is no more install-deps.sh nor the usage of env/bin/python. 
It would be great if the project could be upgraded to run on a fresh Raspbian image, because I'd like to include it in another project. 
Thanks

alonchis, 2018-08-04:
i followed the instructions and i keep getting a "no module named aiy" error.

ndmax, 2018-08-06:
@alonchis The missing aiy module is another issue and can be solved by copying the src/aiy directory into the same directory as whatever directory the script you're trying to use.  So, for example, to run a checkpoint script:

cp -R src/aiy checkpoints

source envbin/activate
(env) checkpoints/check_audio.py

dmitriykovalev, 2018-08-07:
@anuraj1 We just released new SD card image and pushed a lot of commits to GitHub repo, please take a look at new `HACKING.md` file.

ndmax, 2018-08-08:
Still no joy.  I just installed the new SD image, and still receive no response, with 100% google.internal.speech.S3OnePlatform.Assistant 499 errors.

dikkedimi, 2018-08-09:
Here the service is working, but crashes immedately with this error:

`voice-recognizer.service: Failed at step EXEC spawning /home/pi/AIY-projects-python/env/bin/python3: No such file or directory`

dmitriykovalev, 2018-08-10:
@dikkedimi Looks like `voice-recognizer.service` is from the old SD card image, we don't have `virtualenv` (`/home/pi/AIY-projects-python/env/bin/python3`) anymore.

dikkedimi, 2018-08-11:
@dmitriykovalev I've already removed all installed previous versions and followed the instructions by google. Still doesn't work. I've edited the .service file as well, trying different versions found on the forums while searching the received log messages. it's not working with the current instructions. If you can direct me to some correct instructions that work on any raspbian install that'd be more helpful. I'd already read that virtual env wasn't being used anymore but nowhere is there a modified instruction on how to set up the service by yourself. Probably all an effort by Google to make purchasing a Home device easier.

indrora, 2018-08-21:
I'm going to toss my hat into the ring and mention that on a current Raspbian light, the process chokes out because of a file naming issue:

```pi@raspberrypi:~ $ sudo apt-get install aiy-dkms aiy-voicebonnet-routes
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  aiy-io-mcu-firmware aiy-vision-firmware
The following NEW packages will be installed:
  aiy-dkms aiy-io-mcu-firmware aiy-vision-firmware aiy-voicebonnet-routes
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 2,441 kB of archives.
After this operation, 6,901 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-io-mcu-firmware all 1.0-0 [12.5 kB]
Get:2 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-vision-firmware all 1.1-0 [2,404 kB]
Get:3 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-dkms all 1.1 [22.9 kB]
Get:4 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-voicebonnet-routes all 1.0-0 [2,022 B]
Fetched 2,441 kB in 2s (943 kB/s)
Selecting previously unselected package aiy-io-mcu-firmware.
(Reading database ... 67904 files and directories currently installed.)
Preparing to unpack .../aiy-io-mcu-firmware_1.0-0_all.deb ...
Unpacking aiy-io-mcu-firmware (1.0-0) ...
Selecting previously unselected package aiy-vision-firmware.
Preparing to unpack .../aiy-vision-firmware_1.1-0_all.deb ...
Unpacking aiy-vision-firmware (1.1-0) ...
Selecting previously unselected package aiy-dkms.
Preparing to unpack .../archives/aiy-dkms_1.1_all.deb ...
Unpacking aiy-dkms (1.1) ...
Selecting previously unselected package aiy-voicebonnet-routes.
Preparing to unpack .../aiy-voicebonnet-routes_1.0-0_all.deb ...
Unpacking aiy-voicebonnet-routes (1.0-0) ...
Setting up aiy-voicebonnet-routes (1.0-0) ...
Setting up aiy-io-mcu-firmware (1.0-0) ...
ln: failed to create symbolic link '/usr/lib/systemd/system/aiy_mcu_updater.service': No such file or directory
dpkg: error processing package aiy-io-mcu-firmware (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of aiy-dkms:
 aiy-dkms depends on aiy-io-mcu-firmware; however:
  Package aiy-io-mcu-firmware is not configured yet.

dpkg: error processing package aiy-dkms (--configure):
 dependency problems - leaving unconfigured
Setting up aiy-vision-firmware (1.1-0) ...
Errors were encountered while processing:
 aiy-io-mcu-firmware
 aiy-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)```

This also has the effect of breaking a lot of things, and renders my Pi unbootable.

promasu, 2018-09-21:
Just ```mkdir /usr/lib/systemd/system``` and then the installation works

dmitriykovalev, 2018-11-17:
We released updated debian packages and new SD card image, please try to follow updated [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/v20181116/HACKING.md).

closed 2018-11-17 by dmitriykovalev