google/aiyprojects-raspbian #447: THERE ARE NO ASSISTANT AUDIO DATA RESPONCE?

phong31151812, 2018-08-02:
I set up a kit like introduction of v1 with rpi3b everything is ok until i run the demo, it can recognize my voice, but it only return text, no sound audio return, i flash again and use old credential that create long time ago and it's work, but when i flash again and and use a new credential: there no answer sound respond, i check the code and found that there's no data audio return when recognize. Any body have the same issue? please help me.

manoj7410, 2018-08-09:
@phong31151812 Can you try with the latest sdcard image ? Image can be downloaded from : https://github.com/google/aiyprojects-raspbian/releases

cray-, 2018-08-09:
Tested the new 2018-08-03 image on a rpi3b with v1 voicehat. Still no response from the demos. grpc demo correctly recognises commands, everything shows up on myactivity.google.com, check_audio works fine (although my recorded voice sounds very quiet). But not getting any audio responses.

dmitriykovalev, 2018-08-10:
@cray That's weird, when did you setup your credentials? You are right, looks the same as  googlesamples/assistant-sdk-python#274, we'll double check everything.

cray-, 2018-08-11:
@dmitriykovalev I used an existing assistant.json file from a project I created when the original v1 voicekit came out with MagPi in Australia last year. I also flashed a fresh 2018-08-03 image and created a new set of Oauth2 credentials on the same project. Results the same. I have not tried to create a whole new project from scratch.

dmitriykovalev, 2018-08-13:
@cray-  I'm glad it's working now, thank you for the update!

closed 2018-08-16 by dmitriykovalev