google/aiyprojects-raspbian #448: No Response from Assistant API Console Error: S3OnePlatform.Assistant

ndmax, 2018-08-02:
I am unable to get any src/examples/voice demo to respond.  I can push the button or say "Okay Google" (depending on demo) and the button light responds, but there is no audio response from the speaker.  On the console, I am seeing 100% 499 errors from `google.internal.speech.S3OnePlatform.Assistant`

Can anyone help me resolve this issue?

Here's my environment:

Raspberry Pi 2
Google Voice Kit (V1 HAT)
Raspbian Stretch (2018-06-27)
Exec check_audio.py results: "The audio seems to be working."

Here's my  .asoundrc:
```
options snd_rpi_googlevoicehat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}
```

ronmillsap, 2018-08-07:
I have the very same issue

dmitriykovalev, 2018-08-07:
Please try to use the latest SD card image from https://github.com/google/aiyprojects-raspbian/releases.
ALSA config is not required anymore, everything is PulseAudio-based.

ndmax, 2018-08-08:
Still no joy.  I just installed the new SD image, and still receive no response, with 100% google.internal.speech.S3OnePlatform.Assistant 499 errors.

manoj7410, 2018-08-08:
@ndmax Can you please confirm your environment:

Raspberry Pi 2
Google Voice Kit (V1 HAT)
Google AIY image (AIY Kits Release 2018-08-03) from  https://github.com/google/aiyprojects-raspbian/releases

And having assistant.json file in /home/pi ?

ndmax, 2018-08-08:
> Raspberry Pi 2
> Google
> Voice Kit (V1 HAT)
> Google AIY image (AIY Kits Release 2018-08-03) from https://github.com/google/aiyprojects raspbian/releases
> And having assistant.json file in /home/pi ?

@manoj7410 All of the above information is correct.

ronmillsap, 2018-08-08:
I too have created a new SD with version 2018-08-03 and likewise everything works (including checkpoint for audio) except no audio response. Console indicates "vision" response only.

The last line of response reads:
"'command': 'action.devices.commands.VISUAL_OUTPUT'}]}]}}]}"

dmitriykovalev, 2018-08-08:
@ronmillsap Can share the whole response text? I'm not sure what `"'command': 'action.devices.commands.VISUAL_OUTPUT'}]}]}}]}"` means.

ronmillsap, 2018-08-08:
Ok, but it's long.
pi@raspberrypi:~/AIY-projects-python $ src/examples/voice/assistant_library_demo.py
ON_MUTED_CHANGED:
  {'is_muted': False}
/opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
ON_START_FINISHED
Say "OK, Google" then speak, or press Ctrl+C to quit...
ON_DEVICE_ACTION:
  {'inputs': [{'payload': {'commands': [{'execution': [{'params': {'html': '<html> <head><meta charset="UTF-8"> <link href="https://fonts.googleapis.com/css?family=Roboto:400,700" rel="stylesheet"></head> <style>html,body{background:transparent;margin:0}#popout{bottom:0;box-sizing:border-box;font-family:"Roboto",sans-serif;font-size:40px;-webkit-font-smoothing:antialiased;-moz-osx-font-smoothing:grayscale;font-smoothing:antialiased;position:absolute;transform:translateZ(0);transition:opacity 500ms;transition-timing-function:ease-in-out;width:100%}#popout div{font-family:\'Roboto\',arial,sans-serif}.popout-shadow{background-image:linear-gradient(-180deg,rgba(0,0,0,0.0) 0%,rgba(0,0,0,0.8) 100%);height:5.4em;width:100%}.popout-content{background-color:rgba(25,25,25,.97)}.popout-content:hover,.popout-content:focus{outline:none}.popout-asbar:hover,.popout-asbar:focus{outline:none}.inactive{opacity:.4}#carousel-container{left:0;overflow-x:initial;transition:300ms ease-in-out;transform:translateZ(0)}</style> <script>\nwindow.Assistant = window.Assistant || {};var Assistant = window.Assistant;if (Assistant.clear)Assistant.clear();\nAssistant.clear = function(){if (Assistant.cleanup){Assistant.cleanup();}\nwindow.Assistant = {};};</script> <script>\nwindow.Assistant = window.Assistant || {};var Assistant = window.Assistant;\nAssistant.displayTimeoutMs = 20000;\nAssistant.micTimeoutMs = 0;\nAssistant.hideTimerId = undefined;\nAssistant.micTimerId = undefined;\nAssistant.isUsingRelativeIndex = false;\nAssistant.SUGGESTION_PROMPT_ID = \'suggestion_prompt\';\nAssistant.SUGGESTION_PROMPT;\nAssistant.SUGGESTION_CLASS_NAME = \'suggestion\';\nAssistant.POPOUT;\nAssistant.ICON;\nAssistant.cleanup = function(){Assistant.cleanupTimer_();};\nAssistant.updateDisplayTimeoutMs = function(displayTimeoutMs){if (displayTimeoutMs >= 0){Assistant.displayTimeoutMs = displayTimeoutMs;}\n};\nAssistant.updateMicTimeoutMs = function(micTimeoutMs){if (micTimeoutMs >= 0){Assistant.micTimeoutMs = micTimeoutMs;}\n};\nAssistant.updateIsUsingRelativeIndex = function(usingRelativeIndex){Assistant.isUsingRelativeIndex = !!usingRelativeIndex;};\nAssistant.cleanupTimer_ = function(){if (Assistant.hideTimerId){window.clearTimeout(Assistant.hideTimerId);Assistant.hideTimerId = undefined;}\nif (Assistant.micTimerId){window.clearTimeout(Assistant.micTimerId);Assistant.micTimerId = undefined;}\n};\nAssistant.getSuggestionPrompt = function(){if (!Assistant.SUGGESTION_PROMPT){Assistant.SUGGESTION_PROMPT =\ndocument.getElementById(Assistant.SUGGESTION_PROMPT_ID);}\nreturn Assistant.SUGGESTION_PROMPT;};\nAssistant.run = function(){Assistant.POPOUT = document.getElementById(\'popout\');Assistant.ICON = document.getElementsByClassName(\'assistant-icon\')[0];Assistant.keepShowing();};\nAssistant.keepShowing = function(){if (!Assistant.POPOUT){return;}\nAssistant.cleanupTimer_();Assistant.slideUpPopout();if (Assistant.displayTimeoutMs){Assistant.hideTimerId = setTimeout(function(){Assistant.hideTimerId = undefined;Assistant.slideDownPopout();if (typeof Assistant.maybePopulateNotification !== \'undefined\'){Assistant.maybePopulateNotification();}\n},Assistant.displayTimeoutMs);}\nif (Assistant.micTimeoutMs){Assistant.micStartListen();Assistant.micTimerId = setTimeout(function(){Assistant.micTimerId = undefined;Assistant.micStopListen();},Assistant.micTimeoutMs);}\n};\nAssistant.toggleElement = function(ele,display){var target = display ?\'\' :\'none\';if (ele.style.display != target){ele.style.display = target;}\n};\nAssistant.updateSuggestionChips = function(isFirstPage,isLastPage,index){var suggestionEles =\ndocument.getElementsByClassName(Assistant.SUGGESTION_CLASS_NAME);var hasSuggestions = false;for (var i = 0;i < suggestionEles.length;i++){var ele = suggestionEles[i];var suggestStr = ele.textContent.toLowerCase();if (suggestStr == \'previous page\' || suggestStr == \'go back\'){Assistant.toggleElement(ele,!isFirstPage);if (!isFirstPage){hasSuggestions = true;}\n}else if (suggestStr == \'next page\' || suggestStr == \'more\' ||\nsuggestStr == \'show more\'){Assistant.toggleElement(ele,!isLastPage);if (!isLastPage){hasSuggestions = true;}\n}else {hasSuggestions = true;}\n}\nAssistant.toggleElement(Assistant.getSuggestionPrompt(),hasSuggestions);};\nAssistant.micStopListen = function(){Assistant.ICON.classList.add(\'stop\');};\nAssistant.micStartListen = function(){Assistant.ICON.classList.remove(\'stop\');};\nAssistant.slideUpPopout = function(){Assistant.POPOUT.classList.add(\'slide-up\');};\nAssistant.slideDownPopout = function(){Assistant.POPOUT.classList.remove(\'slide-up\');};\nAssistant.close = function(){Assistant.slideDownPopout();Assistant.clear();};\nAssistant.closeWithDelay = function(delayMs){delayMs = Math.max(Number(delayMs),0);setTimeout(Assistant.close,delayMs);};\nAssistant.commandMap = new Map([\n[\'keep_showing\',Assistant.keepShowing],[\'stop_mic\',Assistant.micStopListen],[\'hide\',Assistant.slideDownPopout],[\'close\',Assistant.close],[\'close_with_delay\',Assistant.closeWithDelay],]);\nAssistant.re [... cut ...] Assistant.timeoutMilliSeconds = 30000;</script>  <script>(function(){var displayTimeoutMs=60000;var micTimeoutMs=0;var isUsingRelativeIndex=false;var cardMessages=[{key:\'card_hide\',value:\'{\\n  \\x22event\\x22: {\\n    \\x22cardHide\\x22: true\\n  }\\n}\\n\'}];Assistant.updateIsUsingRelativeIndex(isUsingRelativeIndex);})();</script> </html>'}, 'command': 'action.devices.commands.VISUAL_OUTPUT'}], 'devices': [{'id': '37057F0B6CE93D1E25BD457AA121FE08'}]}]}, 'intent': 'action.devices.EXECUTE'}]}

ronmillsap, 2018-08-08:
Here it is after an "OK Google"
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'what time is it'}
ON_DEVICE_ACTION:
  {'inputs': [{'payload': {'commands': [{'execution': [{'params': {'html': '<html> <head><meta charset="UTF-8"> <link href="https://fonts.googleapis.com/css?family=Roboto:400,700" rel="stylesheet"></head> <style>html,body{background:transparent;margin:0}#popout{bottom:0;box-sizing:border-box;font-family:"Roboto",sans-serif;font-size:40px;-webkit-font-smoothing:antialiased;-moz-osx-font-smoothing:grayscale;font-smoothing:antialiased;position:absolute;transform:translateZ(0);transition:opacity 500ms;transition-timing-function:ease-in-out;width:100%}#popout div{font-family:\'Roboto\',arial,sans-serif}.popout-shadow{background-image:linear-gradient(-180deg,rgba(0,0,0,0.0) 0%,rgba(0,0,0,0.8) 100%);height:5.4em;width:100%}.popout-content{background-color:rgba(25,25,25,.97)}.popout-content:hover,.popout-content:focus{outline:none}.popout-asbar:hover,.popout-asbar:focus{outline:none}.inactive{opacity:.4}#carousel-container{left:0;overflow-x:initial;transition:300ms ease-in-out;transform:translateZ(0)}</style> <script>\nwindow.Assistant = window.Assistant || {};var Assistant = window.Assistant;if (Assistant.clear)Assistant.clear();\nAssistant.clear = function(){if (Assistant.cleanup){Assistant.cleanup();}\nwindow.Assistant = {};};</script> <script>\nwindow.Assistant = window.Assistant || {};var Assistant = window.Assistant;\nAssistant.displayTimeoutMs = 20000;\nAssistant.micTimeoutMs = 0;\nAssistant.hideTimerId = undefined;\nAssistant.micTimerId = undefined;\nAssistant.isUsingRelativeIndex = false;\nAssistant.SUGGESTION_PROMPT_ID = \'suggestion_prompt\';\nAssistant.SUGGESTION_PROMPT;\nAssistant.SUGGESTION_CLASS_NAME = \'suggestion\';\nAssistant.POPOUT;\nAssistant.ICON;\nAssistant.cleanup = function(){Assistant.cleanupTimer_();};\nAssistant.updateDisplayTimeoutMs = function(displayTimeoutMs){if (displayTimeoutMs >= 0){Assistant.displayTimeoutMs = displayTimeoutMs;}\n};\nAssistant.updateMicTimeoutMs = function(micTimeoutMs){if (micTimeoutMs >= 0){Assistant.micTimeoutMs = micTimeoutMs;}\n};\nAssistant.updateIsUsingRelativeIndex = function(usingRelativeIndex){Assistant.isUsingRelativeIndex = !!usingRelativeIndex;};\nAssistant.cleanupTimer_ = function(){if (Assistant.hideTimerId){window.clearTimeout(Assistant.hideTimerId);Assistant.hideTimerId = undefined;}\nif (Assistant.micTimerId){window.clearTimeout(Assistant.micTimerId);Assistant.micTimerId = undefined;}\n};\nAssistant.getSuggestionPrompt = function(){if (!Assistant.SUGGESTION_PROMPT){Assistant.SUGGESTION_PROMPT =\ndocument.getElementById(Assistant.SUGGESTION_PROMPT_ID);}\nreturn Assistant.SUGGESTION_PROMPT;};\nAssistant.run = function(){Assistant.POPOUT = document.getElementById(\'popout\');Assistant.ICON = document.getElementsByClassName(\'assistant-icon\')[0];Assistant.keepShowing();};\nAssistant.keepShowing = function(){if (!Assistant.POPOUT){return;}\nAssistant.cleanupTimer_();Assistant.slideUpPopout();if (Assistant.displayTimeoutMs){Assistant.hideTimerId = setTimeout(function(){Assistant.hideTimerId = undefined;Assistant.slideDownPopout();if (typeof Assistant.maybePopulateNotification !== \'undefined\'){Assistant.maybePopulateNotification();}\n},Assistant.displayTimeoutMs);}\nif (Assistant.micTimeoutMs){Assistant.micStartListen();Assistant.micTimerId = setTimeout(function(){Assistant.micTimerId = undefined;Assistant.micStopListen();},Assistant.micTimeoutMs);}\n};\nAssistant.toggleElement = function(ele,display){var target = display ?\'\' :\'none\';if (ele.style.display != target){ele.style.display = target;}\n};\nAssistant.updateSuggestionChips = function(isFirstPage,isLastPage,index){var suggestionEles =\ndocument.getElementsByClassName(Assistant.SUGGESTION_CLASS_NAME);var hasSuggestions = false;for (var i = 0;i < suggestionEles.length;i++){var ele = suggestionEles[i];var suggestStr = ele.textContent.toLowerCase();if (suggestStr == \'previous page\' || suggestStr == \'go back\'){Assistant.toggleElement(ele,!isFirstPage);if (!isFirstPage){hasSuggestions = true;}\n}else if (suggestStr == \'next page\' || suggestStr == \'more\' ||\nsuggestStr == \'show more\'){Assistant.toggleElement(ele,!isLastPage);if (!isLastPage){hasSuggestions = true;}\n}else {hasSuggestions = true;}\n}\nAssistant.toggleElement(Assistant.getSuggestionPrompt(),hasSuggestions);};\nAssistant.micStopListen = function(){Assistant.ICON.classList.add(\'stop\');};\nAssistant.micStartListen = function(){Assistant.ICON.classList.remove(\'stop\');};\nAssistant.slideUpPopout = function(){Assistant.POPOUT.classList.add(\'slide-up\');};\nAssistant.slideDownPopout = function(){Assistant.POPOUT.classList.remove(\'slide-up\');};\nAssistant.close = function(){Assistant.slideDownPopout();Assistant.clear();};\nAssistant.closeWithDelay = function(delayMs){delayMs = Math.max(Number(delayMs),0);setTimeout(Assistant.close,delayMs);};\nAssistant.commandMap = new Map([\n[\'keep_showing\',Assistant.keepShowing],[\'stop_mic\',Assistant.micStopListen],[\'hide\',Assistant.slideDownPopout],[\'close\',Assistant.close],[\'close_with_delay\',Assistant.closeWithDelay],]);\nAssistant.registerCommand = function(commandName,commandFunc){commandName = commandName.toLowerCase();if (!Assistant.commandMap.has(commandName))\nAssistant.commandMap.set(commandName,commandFunc);};\nAssistant.runCmd = function(cmdName,arg1,arg2){cmdName = cmdName.toLowerCase();var cmdFu [... cut ...] utMilliSeconds = 30000;</script>  <script>(function(){var displayTimeoutMs=60000;var micTimeoutMs=0;var isUsingRelativeIndex=false;var cardMessages=[{key:\'card_hide\',value:\'{\\n  \\x22event\\x22: {\\n    \\x22cardHide\\x22: true\\n  }\\n}\\n\'}];Assistant.updateIsUsingRelativeIndex(isUsingRelativeIndex);})();</script> </html>'}, 'command': 'action.devices.commands.VISUAL_OUTPUT'}], 'devices': [{'id': '37057F0B6CE93D1E25BD457AA121FE08'}]}]}, 'intent': 'action.devices.EXECUTE'}]}
ON_NO_RESPONSE

manoj7410, 2018-08-09:
@ronmillsap Did you try to run assistant_grpc_demo too ? Does it work ?

dmitriykovalev, 2018-08-09:
@ronmillsap Is there any voice response when these long messages are printed? Did you follow step "Give permissions to the Google Assistant API" from https://aiyprojects.withgoogle.com/voice/?

ronmillsap, 2018-08-09:
Dimitri, sorry I missed your first question. There is no sound whatsoever with the response however the light does come on just like when there IS a response, just without any sound.
I am curious about running grpc so I will swap out the button for the led and see if grpc works or not.

ronmillsap, 2018-08-09:
Ok, set it up the button, all check points worked fine, started grpc and the light started blinking slowly (ready). Pressed the button and the following occured: however, this time the button light started blinking quick short blinks (pulse=thinking) as well as the slow blinks.

Build info: Sat Aug 4 00:20:37 UTC 2018 @ 51293f3
Dev terminal is ready! See the demos in 'src/examples' to get started.
Note: the MagPi Essentials are not up-to-date. See https://goo.gl/8hEcfo
pi@raspberrypi:~/AIY-projects-python $ src/examples/voice/assistant_grpc_demo.py/opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
[2018-08-09 10:01:20,658] INFO:recorder:started recording
Press the button and speak
Listening...
[2018-08-09 10:01:48,349] INFO:speech:transcript: what
[2018-08-09 10:01:48,355] INFO:speech:transcript: what's
[2018-08-09 10:01:48,357] INFO:speech:transcript: what time
[2018-08-09 10:01:48,461] INFO:speech:transcript: what time is
[2018-08-09 10:01:48,614] INFO:speech:transcript: what time is it
[2018-08-09 10:01:48,797] INFO:speech:transcript: what time  is it
[2018-08-09 10:01:49,159] INFO:speech:transcript: what time is  it
[2018-08-09 10:01:49,198] INFO:speech:transcript: what time is it
[2018-08-09 10:01:50,633] INFO:speech:event_type: 1
[2018-08-09 10:01:50,699] INFO:speech:transcript: what time is it
You said " what time is it "
Press the button and speak
Listening...
[2018-08-09 10:03:09,648] INFO:speech:transcript: ok
[2018-08-09 10:03:09,709] INFO:speech:transcript: Okay Goo
[2018-08-09 10:03:10,016] INFO:speech:transcript: OK Google
[2018-08-09 10:03:10,658] INFO:speech:transcript: OK Google
[2018-08-09 10:03:18,065] INFO:speech:transcript: OK Google  what
[2018-08-09 10:03:18,206] INFO:speech:transcript: OK Google  what time
[2018-08-09 10:03:18,625] INFO:speech:transcript: OK Google  what time is
[2018-08-09 10:03:18,699] INFO:speech:transcript: OK Google  what time is it
[2018-08-09 10:03:18,702] INFO:speech:transcript: OK Google what  time is it
[2018-08-09 10:03:18,753] INFO:speech:transcript: OK Google what time  is it
[2018-08-09 10:03:18,974] INFO:speech:transcript: OK Google what time is  it
[2018-08-09 10:03:18,978] INFO:speech:transcript: OK Google what time is it
[2018-08-09 10:03:19,802] INFO:speech:event_type: 1
[2018-08-09 10:03:19,860] INFO:speech:transcript: OK Google what time is it
You said " OK Google what time is it "
Press the button and speak

dmitriykovalev, 2018-08-10:
@ronmillsap Interesting, not sure what's happening here, output looks good. Just to double check, you said that all checkpoints passed, right? So you were able to record then listen to your voice during sound check procedure?

ronmillsap, 2018-08-10:
Yes Dimitri, today I walked through the instructions again. In troubleshooting, it suggested to repeat step 7 which is to get the oAth2.0 client. So I tried new credentials but got the very same results. 
I do wonder what the consequences of the warning about "this channel is already in use" as that warning is always shown when I start up?

dmitriykovalev, 2018-08-13:
@ronmillsap Can you please look at the similar issue #447? Looks like everything was resolved there by just creating a new project. Regarding `this channel is already in use`, that shouldn't be a problem but we'll try to figure out and fix this.

ronmillsap, 2018-08-13:
Yes I will give it a go. Thanks for not giving up!

ronmillsap, 2018-08-14:
ok Dmitriy, i was able to get everything working properly by setting up a new project. not sure EXACTLY what that means or why it works but all is working. I'll take that. Thank you for all the help!

dmitriykovalev, 2018-08-15:
@ronmillsap Great! It's actually more related to [Google Assistant Library](https://github.com/googlesamples/assistant-sdk-python) than to our project :)

dmitriykovalev, 2018-10-05:
Looks like this problem doesn't happen anymore.

closed 2018-10-05 by dmitriykovalev