google/aiyprojects-raspbian #449: Duplicate alert events

mpember, 2018-08-02:
Whenever a timer or alarm is set, the following occurs when the timer / alarm goes off:

A **ON_ALERT_STARTED** event is followed by an immediate **ON_ALERT_FINISHED** event, followed by a second **ON_ALERT_STARTED** event.

A second **ON_ALERT_FINISHED** event is not triggered until you wake the assistant, to stop the alarm.

Is this expected?  Is this because the API generates multiple, redundant events?

dmitriykovalev, 2018-08-16:
This issue probably should be reported to https://github.com/googlesamples/assistant-sdk-python/issues  We are actually using an old version (0.1.0) but going to upgrade soon.

dmitriykovalev, 2018-11-17:
We use the latest Assistant Library in [the new SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116).

closed 2018-11-17 by dmitriykovalev