google/aiyprojects-raspbian #454: Error while installing aiy-io-mcu-firmware and aiy-dkms

nickdex, 2018-08-09:
I tried the out the (updated) steps on `Hacking.md` on fresh raspbian install, encountered some errors. Pasting the output -
```Console
➜ sudo apt-get install aiy-dkms aiy-voicebonnet-soundcard-dkms aiy-voicebonnet-routes
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  aiy-io-mcu-firmware aiy-vision-firmware dkms raspberrypi-kernel-headers
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  aiy-dkms aiy-io-mcu-firmware aiy-vision-firmware aiy-voicebonnet-routes
  aiy-voicebonnet-soundcard-dkms dkms raspberrypi-kernel-headers
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 18.7 MB of archives.
After this operation, 112 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-kernel-headers armhf 1.20180703-1 [16.1 MB]
Get:3 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-io-mcu-firmware all 1.0-0 [12.5 kB]
Get:4 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-vision-firmware all 1.1-0 [2,404 kB]
Get:2 http://mirror.ossplanet.net/raspbian/raspbian stretch/main armhf dkms all 2.3-2 [74.8 kB]
Get:5 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-dkms all 1.1 [22.9 kB]
Get:6 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-voicebonnet-soundcard-dkms all 1.1 [32.8 kB]
Get:7 https://dl.google.com/aiyprojects/deb stable/main armhf aiy-voicebonnet-routes all 1.0-0 [2,022 B]
Fetched 18.7 MB in 16s (1,125 kB/s)
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 38558 files and directories currently installed.)
Preparing to unpack .../0-raspberrypi-kernel-headers_1.20180703-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20180703-1) ...
Selecting previously unselected package aiy-io-mcu-firmware.
Preparing to unpack .../1-aiy-io-mcu-firmware_1.0-0_all.deb ...
Unpacking aiy-io-mcu-firmware (1.0-0) ...
Selecting previously unselected package aiy-vision-firmware.
Preparing to unpack .../2-aiy-vision-firmware_1.1-0_all.deb ...
Unpacking aiy-vision-firmware (1.1-0) ...
Selecting previously unselected package dkms.
Preparing to unpack .../3-dkms_2.3-2_all.deb ...
Unpacking dkms (2.3-2) ...
Selecting previously unselected package aiy-dkms.
Preparing to unpack .../4-aiy-dkms_1.1_all.deb ...
Unpacking aiy-dkms (1.1) ...
Selecting previously unselected package aiy-voicebonnet-soundcard-dkms.
Preparing to unpack .../5-aiy-voicebonnet-soundcard-dkms_1.1_all.deb ...
Unpacking aiy-voicebonnet-soundcard-dkms (1.1) ...
Selecting previously unselected package aiy-voicebonnet-routes.
Preparing to unpack .../6-aiy-voicebonnet-routes_1.0-0_all.deb ...
Unpacking aiy-voicebonnet-routes (1.0-0) ...
Setting up raspberrypi-kernel-headers (1.20180703-1) ...
Setting up aiy-io-mcu-firmware (1.0-0) ...
ln: failed to create symbolic link '/usr/lib/systemd/system/aiy_mcu_updater.service': No such file or directory
dpkg: error processing package aiy-io-mcu-firmware (--configure):
 subprocess installed post-installation script returned error exit status 1
Setting up dkms (2.3-2) ...
dpkg: dependency problems prevent configuration of aiy-dkms:
 aiy-dkms depends on aiy-io-mcu-firmware; however:
  Package aiy-io-mcu-firmware is not configured yet.

dpkg: error processing package aiy-dkms (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.7.6.1-2) ...
Setting up aiy-vision-firmware (1.1-0) ...
Setting up aiy-voicebonnet-soundcard-dkms (1.1) ...
Loading new aiy-voicebonnet-soundcard-1.1 DKMS files...
It is likely that 4.14.52-v7+ belongs to a chroot's host
Building for 4.14.52+ and 4.14.52-v7+
Building for architecture armhf
Building initial module for 4.14.52+
Done.

snd-aiy-voicebonnet:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.52+/kernel/sound/drivers/

rt5645.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.52+/kernel/sound/soc/codecs/

rl6231.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.52+/kernel/sound/soc/codecs/

depmod....

DKMS: install completed.
Building initial module for 4.14.52-v7+
Done.

snd-aiy-voicebonnet:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.52-v7+/kernel/sound/drivers/

rt5645.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.52-v7+/kernel/sound/soc/codecs/

rl6231.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.14.52-v7+/kernel/sound/soc/codecs/

depmod....

DKMS: install completed.
Setting up aiy-voicebonnet-routes (1.0-0) ...
Errors were encountered while processing:
 aiy-io-mcu-firmware
 aiy-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

****
```

manoj7410, 2018-08-10:
@nickdex Which voice kit do you have ? I just finished all the steps on Voice V1 and didn't faced any errors till "arecord and aplay".

jessecha, 2018-08-10:
I have the same problem.

nickdex, 2018-08-11:
@manoj7410 I'm using Voice Kit V1 as well. Just to be sure I didn't misinterpret or left out anything, can you please write down your steps (which I can copy, paste and run)

dkv, 2018-08-11:
`sudo mkdir /usr/lib/systemd/system` seems to resolve the issue.
Then you can run: `sudo dpkg --configure aiy-io-mcu-firmware`

petetum, 2018-11-05:
First error while configuring aiy-io-mcu-firmware:
failed to create symbolic link '/usr/lib/systemd/system/aiy_mcu_updater.service': No such file or directory

Second error wile configuring aiy-dkms:
dependency problems - leaving unconfigured
This package needs aiy-io-mcu-firmware to be configured first.

To resolve issue create the directory then configure both packages:

sudo mkdir /usr/lib/systemd/system
sudo dpkg --configure aiy-io-mcu-firmware
sudo dpkg --configure aiy-dkms