google/aiyprojects-raspbian #457: aiy-python-wheels: too many levels of symbolic links error during installation

jessecha, 2018-08-10:
As the title suggests, when I try to install aiy-python-wheels, I get an dpkg error indicating too many levels of symbolic links.

dmitriykovalev, 2018-08-10:
Can you please copy error message here?

jessecha, 2018-08-11:
I ended up ditching the ubuntu mate installation attempt, and I do not have a copy of the error message.

dmitriykovalev, 2018-08-13:
Oh, Ubuntu Mate, got it. That's an important information you should've put in the description.

sagescolorado, 2018-09-20:
same issue -  i'm using Ubuntu made 16.04 on an rpi3

I get the same error every time I install anything now.

Error message:
Setting up aiy-python-wheels (1.0-0) ...
Failed to execute operation: Too many levels of symbolic links
dpkg: error processing package aiy-python-wheels (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 aiy-python-wheels

dmitriykovalev, 2018-11-17:
`aiy-python-wheels` debian package is already fixed.

closed 2018-11-17 by dmitriykovalev