google/aiyprojects-raspbian #460: SSD Mobilenet Binaryproto run Object Detection with bounding boxes live in AIY Camera

JakeBMalis, 2018-08-13:
I managed to retrain the Mobilenet v1 SSD (256 x 265, 0.125 multiplier) for my deck of cards. I converted the .pb file into a .binaryproto. I need help modifying the object_detection.py file to have the bounding boxes in the live camera feed. Does anyone happen to have code to run it live in the camera preview?

Thanks!

giacomobartoli, 2018-08-14:
@JakeBMalis I did it. Take a look at my repo: https://github.com/giacomobartoli/vision-kit/tree/master/pikachu-detector

giacomobartoli, 2018-09-26:
Again, take a look at my repository. You need those two libraries + a protobinary file.

JakeMalis, 2018-09-26:
I can't figure out how to make it work real-time. Even following along with your code.

giacomobartoli, 2018-09-27:
Open an issue into my repo and I'll be glad to help you.

closed 2021-01-15 by JakeMalis