google/aiyprojects-raspbian #464: can't use inside virtualenvironment because of numpy

indrora, 2018-08-21:
The voice hat driver calls some numpy to make the volume happen. Unfortunately, this is neither described in `setup.py` nor is it, as a result, usable inside a virtual environment.

what I'm doing:

in a virtualenv, I've got pip reading the package as is described in HACKING.md 

I've then done

```import aiy.voicehat as hat```

what should happen:

the module gets imported. World peace drifts over the sky like a soft, warm embrace. 

what really happens:

either

* numpy isn't installed, leading to the import failing
* numpy can't load libf77blas.so.3 because it can't load it from the system.

I've determined that commit 8efe47ded7d9f0974b29d6c9a407fe1997ed2adb introduced the undocumented numpy dependency.

manoj7410, 2018-08-21:
@indrora Is virtualenv a necessity for your objective ?

indrora, 2018-08-21:
Yes, it is. Virtualenv is a hard dependency of Tox (unit tests) and basic python cleanliness. It's vaguely recommended by the PSF and generally considered a best practice. It separates you from any distribution-specific issues or highly out of date packages that may be lingering from slower distros (e.g. CentOS, etc).

A workaround is to generate a virtualenv with `--system-site-packages` as I discovered. Numpy will load from system packages happily, but this does not fix the underlying issue. I'll update the title to better reflect it.

dmitriykovalev, 2018-08-23:
`numpy` shouldn't be a dependency, will try to get rid of it.

dmitriykovalev, 2018-11-17:
Fixed, dependency was removed from the code.

closed 2018-11-17 by dmitriykovalev