google/aiyprojects-raspbian #465: Bootstrapping cloudspeech example with a custom speech-to-text model

gqoew, 2018-08-21:
Hi,

I am trying to bootstrap the repository examples to create a script that does the following:

1. Wait for user to push the button to start speaking.
2. Once button is pushed, start recording the user and stop when he/she has stopped speaking.
3. Transform the audio record in text with a custom model instead of the cloudspeech API.

From what I understood, it is quite seems similar to the [cloudspeech_demo.py](https://github.com/google/aiyprojects-raspbian/blob/93a6306438492464c25916b6b2f53d8e4750579b/src/examples/voice/cloudspeech_demo.py#L37) example. It seems I need to replace `recognizer = aiy.cloudspeech.get_recognizer()` and `text = recognizer.recognize()` by my own model.

I don't really understand how the cloudspeech recognizer works, how it is getting the audio, how it knows when to stop recording, and if it is doing other processing jobs. My current speech-to-text model takes as input a `.wav` file and returns the corresponding text. I thought about using [`aiy.audio.record_to_wave`](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/audio.py#L88) after button is pushed but it seems we can only use this function with a given `duration` parameter which is not very convenient in my case.

Do you have any recommendation to help me achieving the steps from above? Thanks 👍

manoj7410, 2018-08-30:
@gqoew if you look at the constructor in:

 `class _WaveDump(object):
    """A processor that saves recorded audio to a wave file."""

    def __init__(self, filepath, duration):
        self._wave = wave.open(filepath, 'wb')
        self._wave.setnchannels(1)
        self._wave.setsampwidth(2)
        self._wave.setframerate(16000)
        self._bytes = 0
        self._bytes_limit = int(duration * 16000) * 1 * 2`

Here "duration" is being used to decide the memory limit. record_to_wave is not a feasible solution for your problem. I am looking into the code of "class _CloudSpeechRecognizer(object):" 

Will get back to you as soon as I find something helpful.

gqoew, 2018-09-10:
@manoj7410 👌

dmitriykovalev, 2018-09-10:
Looks like the problem is solved.

closed 2018-09-10 by dmitriykovalev