google/aiyprojects-raspbian #466: Voice Hat is not working after install on fresh raspbian (HACKING.md)

gqoew, 2018-08-22:
Hi,

I am following new [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) for the Voice Kit on a fresh raspbian (RPI3 + Voice Hat), I did all the steps successfully until that `arecord -f cd test.wav` produced me an audio file without sound and `aplay test.wav` did not play any sound.

So I runned [check_audio.py](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/checkpoints/check_audio.py), it had the [ERROR_NO_AIY_SOUND_CARDS](https://github.com/google/aiyprojects-raspbian/blob/93a6306438492464c25916b6b2f53d8e4750579b/checkpoints/check_audio.py#L49) error. After looking in the repository issues, I fixed by adding `dtoverlay=googlevoicehat-soundcard` to `/boot/config.txt` (#339)

Then I got the [ERROR_NOT_A_FIRST_SOUND_CARD](https://github.com/google/aiyprojects-raspbian/blob/93a6306438492464c25916b6b2f53d8e4750579b/checkpoints/check_audio.py#L54) that I've fixed with commenting out `dtparam=audio=on` (#451).

Now I have no errors when running [check_audio.py](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/checkpoints/check_audio.py), but I still cannot hear the speaker or record audio... Everything seems also wired properly.

Do you know what could have gone wrong? Thanks!

dmitriykovalev, 2018-08-22:
`check_audio.py` runs microphone and  speaker tests as well. If you have no errors, does it mean that these tests work for you?

manoj7410, 2018-08-29:
@gqoew Can you share the output of lsmod right after booting ?

dmitriykovalev, 2018-11-17:
@gqoew please reopen if that is still a problem for you.

closed 2018-11-17 by dmitriykovalev

phaniteja1, 2018-12-21:
I still have this issue. @gqoew Were you able to fix it? Thanks

phaniteja1, 2018-12-21:
@dmitriykovalev I have the exact same issue. Can you please help. Thanks

Whytehorse, 2019-05-10:
Same prob here. lsmod shows:
Module                  Size  Used by
bnep                   20480  2
hci_uart               36864  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             368640  24 hci_uart,bnep,btbcm
ecdh_generic           28672  1 bluetooth
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              573440  1 brcmfmac
rfkill                 28672  6 bluetooth,cfg80211
snd_bcm2835            32768  0
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  3 snd_timer,snd_bcm2835,snd_pcm
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  425984  42