google/aiyprojects-raspbian #468: Long delays when play and recording audio on the Voice Kit.

cscottlamb, 2018-08-25:
When running the stock image 2018-08-03 on the voice kit I'm seeing excessively long delays playing and recording audio.  Over two minutes to run the check_audio.py script.

pi@raspberrypi:~/AIY-projects-python/checkpoints $ time echo -e "y\n\ny\n" | ./check_audio.py
You have Voice Bonnet (v2) installed at index 0!
Playing a test sound...
Did you hear the test sound? (y/n) When you are ready, press Enter and say "Testing, 1 2 3"...Recording for 3 seconds...
Playing back recorded audio...
Did you hear your own voice? (y/n) AIY sound card seems to be working!
Press Enter to close...
real    2m18.515s
user    0m3.336s
sys     0m0.303s

dmitriykovalev, 2018-08-28:
Which Pi board do you use? Can please also provide `dmesg` output after running your test?

cscottlamb, 2018-08-29:
Thanks for your help Dmitriy! I'm using the Pi Zero W that came with the kit.

Here is the dmesg output: [dmesg.txt](https://github.com/google/aiyprojects-raspbian/files/2330378/dmesg.txt)

The play back and record do end up working. Though I could never give it a voice command because I don't know when its 'really' starting the record. Actually I don't have that part working yet either, first things first. :-)