google/aiyprojects-raspbian #472: AIY Voice - Too Many Requests

CarlP66, 2018-08-30:
I have been using the AIY Voice kit for a couple of weeks, and all was going well - initially just with the assistant_library_demo, and more recently experimenting with local commands.
Then today, literally mid-conversation it completely died. I now get error message "429 Client Error: Too Many Requests for url: https://embeddedassistant.googleapis.com...." (full version below) every time I run any of the demo scripts. I don't believe I changed anything since it was working, and my useage has not been excessive, so it very much looks like an issue at google's end.

Is there a more formal way to raise a support ticket, or is all support through git?

Thanks in advance. The full text of the error is below:

pi@raspberrypi:~/AIY-projects-python $ src/examples/voice/assistant_library_demo.py
Traceback (most recent call last):
  File "src/examples/voice/assistant_library_demo.py", line 74, in <module>
    main()
  File "src/examples/voice/assistant_library_demo.py", line 68, in main
    with Assistant(credentials) as assistant:
  File "/opt/aiy/projects-python/src/aiy/assistant/library.py", line 33, in __init__
    self._model_id = device_helpers.register_model_id(credentials)
  File "/opt/aiy/projects-python/src/aiy/assistant/device_helpers.py", line 93, in register_model_id
    r.raise_for_status()
  File "/usr/local/lib/python3.5/dist-packages/requests/models.py", line 928, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 429 Client Error: Too Many Requests for url: https://embeddedassistant.googleapis.com/v1alpha2/projects/voice-assistant-213123/deviceModels/voice-assistant-213123-voice-kit
pi@raspberrypi:~/AIY-projects-python $

CarlP66, 2018-08-31:
The issue resolved itself the next day, so all is good for now. I am assuming it was a glitch at google's end.

closed 2018-08-31 by CarlP66

sjesusma, 2018-11-01:
Which is the quota?
I have had the same problem sometimes / days.

dmitriykovalev, 2018-11-04:
@sjesusma You can probably find this at https://console.cloud.google.com/

sjesusma, 2018-11-18:
Thanks @dmitriykovalev, my problem is in the quotas on Google Assistant API (max. 500 requests every 24h). When someone is developing, sometimes are required a lot more. I have asked to Google for an extension of this quota, I hope they accept it. :)

dmitriykovalev, 2018-11-18:
@sjesusma Sounds good!