google/aiyprojects-raspbian #473: AIY Voice: any plans for send_text_query() API in Assistant?

jfatula, 2018-08-30:
I've just begun working with VoiceKit (V1) on a Pi3  using the August 3rd 2018 AIY projects image (aiyprojects-2018-08-03.img.xz).  I soon found that the send_text_query() API call documented for the [google.assistant.library.Assistant public release](https://developers.google.com/assistant/sdk/reference/library/python/) isn't present in the aiyprojects build:
```
self._assistant.version: {'core': '107430'}
dir(self._assistant):
 ['__call_', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__',
 '__enter__', '__eq__', '__exit__', '__format__', '__ge__', '__getattribute__',
 '__gt__', '__hash__', '__init__', '__le__', '__lt__', '__module__', '__ne__',
 '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__',
 '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_credentials', 
'_credentials_refresher', '_event_callback', '_event_queue', '_inst', '_lib', 
'_libc', '_load_lib', '_model_id', '_set_credentials',
 'device_id', 'set_mic_mute', 'start', 'start_conversation', 'stop_conversation', 'version']
```
Are there plans to include it in a future release?
Many thanks!

dmitriykovalev, 2018-09-04:
@jfatula Our assistant library version is pretty old, we have plans to update it for the next SD card image release (we don't have exact date yet). You can try to install new library version yourself and see if it's working.

jfatula, 2018-09-10:
@dmitriykovalev Thanks! Using the default August 3rd 2018 AIY projects image,  I ran
`sudo pip3 install --upgrade google.assistant-library` to upgrade the library, which now reports its version as {'package': '1.0.0', 'core': 120038}.  This version does include the send_text_query() API call and preliminary testing shows the call to work as documented, and so far I haven't encountered other issues from using the upgraded assistant library.

dmitriykovalev, 2018-09-10:
@jfatula Perfect!

closed 2018-09-10 by dmitriykovalev