google/aiyprojects-raspbian #474: AIY vision kit with a different camera

jessecha, 2018-08-31:
Is it possible to use the vision board with a different camera?
I know the V1 is not supported, but is it possible to create a generic driver for the vision bonnet? I would like to use a 360 camera.

dmitriykovalev, 2018-08-31:
Pi Camera is fully managed by Pi's driver, VisionBonnet just intercepts raw image frames. There is a crypto chip on Pi Camera, so Pi's driver will not work with any custom camera.

jessecha, 2018-08-31:
If this is the case, why doesnt the vision bonnet work with a v1 camera? I was specifically looking at using a kogeto dot camera for the vision bonnet

dmitriykovalev, 2018-09-01:
v1 and v2 have completely different image sensors (OmniVision vs Sony) and our internal imaging pipeline on Myriad is optimized for v2. Why don't you use v2 with Kogeto Dot?

closed 2018-11-16 by dmitriykovalev