google/aiyprojects-raspbian #475: Click/pop noise at end of aiy.audio.say() playback in 2018-08-03 image

TM-hub, 2018-09-02:
Seems to be due to the <volume level='xx'> modifier in say() (tts.py)
To recreate, from the command line:

**Click version**
`pico2wave -w ~/temp.wav -l en-US "<volume level='40'>hello</volume>"
aplay  temp.wav`

**No click version**
`pico2wave -w ~/temp.wav -l en-US "hello"
aplay  temp.wav`

The click version above has the same result (makes a click/pop) on both 2018-08-03 and 2018-04-13 images.

mikeyroth, 2018-09-23:
I also have a pop at the end when using aiy.audio.say()

f4n, 2018-11-09:
Same here, some workarounds available?

TheArduinist, 2018-11-11:
I also have the same problem. By using this code:

`cmd = 'pico2wave -w %s -l en-US "<volume level=\'40\'>hello</volume>"' % ('temp.wav')`
`subprocess.check_call(cmd, shell=True)`
`cmd = 'aplay %s' % ('temp.wav')`
`subprocess.check_call(cmd, shell=True)`

Which basically explicitly uses the command @TM-hub used, the pop was avoided in my case. I guess that means the problem might lie in the way tts.py issues the command? I'm not sure.

dmitriykovalev, 2018-11-12:
That's already fixed in our internal codebase, we'll publish to GitHub soon. That's the new code:

```python
import os
import subprocess
import tempfile

TMP_DIR = '/run/user/%d' % os.getuid()

def say(text, lang='en-US', volume=60, pitch=130, speed=100, device='default'):
    data = "<volume level='%d'><pitch level='%d'><speed level='%d'>%s</speed></pitch></volume>" % \
        (volume, pitch, speed, text)
    with tempfile.NamedTemporaryFile(suffix='.wav', dir=TMP_DIR) as file:
       cmd = 'pico2wave --wave %s --lang %s "%s" && aplay -q -D %s %s' % (file.name, lang, data, device, file.name)
       subprocess.check_call(cmd, shell=True)

```

dmitriykovalev, 2018-11-17:
Fixed, you can try new SD card image: https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116

closed 2018-11-17 by dmitriykovalev