google/aiyprojects-raspbian #476: Useing Smartphone as monitor stream video is not visible

algila, 2018-09-02:
Hi, 
Connecting the kit to an Android SmartPhone via SSH I am fully able to see and control the terminal (having both on the same WiFi net).
Using VNC viewer I am also able to see the AIY kit desktop, but running for instance `./face_detection_camera.py `
I am not able to see the stream video on my mobile. I am only able to see the terminal information.

The stream video is visible on the monitor when it is physically connected to the AIY kit, but it is not visible on the SmartPhone connected via WiFi using VNC viewer.

Should be a nice feature to see what the kit camera is looking on the SmartPhone. 
Do you know why ?
Any way to have it ?

dmitriykovalev, 2018-09-04:
@algila This is a correct behavior, video from the camera goes directly to the HDMI output and bypasses any desktop compositing. There is a way though to watch camera video in the browser, but it's still experimental, check the [source code](https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/src/aiy/vision/streaming). Video steaming is enabled by default for JoyDemo, so just point your browser to  http://raspberrypi.local:4664/

algila, 2018-09-04:
Many thanks.

About the Video steaming enabled by default for JoyDemo using Mozilla firefox in Ubuntu 18.04 
I see it and with a very small latency. To me it is very good considering also that I'm using my samsung table as WiFi rooter between AIY kit and my desktop PC.

About the source code, may you drive me a little bit more in details about how I need to change face detector to get streaming video on web ?

dmitriykovalev, 2018-09-05:
I guess you can just go through the source code of [JoyDemo](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/joy/joy_detection_demo.py) and see how `StreamingServer` and  `InferenceData` classes are used. You can update face detector example in the similar way. This is still experimental and could be changed in the future.

virtualjmills, 2018-09-08:
Regarding: https://github.com/google/aiyprojects-raspbian/issues/476#issuecomment-418213773

Alternatively, you can reconfigure X11 VNC Server to snoop the direct render layers as well, with a slight overall performance penalty:

VNC Server -> [=] Menu -> Options -> Troubleshooting...

Optimize Screen Capture
[X] Enable Experimental Direct Capture Mode

[OK]

Then disconnect and reconnect your VNC Viewer session.

This change persists properly through reboots.

dmitriykovalev, 2018-09-10:
@virtualjmills Thanks for sharing this info!

algila, 2018-09-11:
Many thanks @virtualjmills it is working quite easily.  
Unfortunately the refresh latency is very high around 1 or 2 second. Is it possible to reduce it with some settings ?