google/aiyprojects-raspbian #478: Unable to autoboot Google AIY voice kit V1 getting folowing error

JITESHNAIR01, 2018-09-02:
● assist.service - Assist @ reboot
   Loaded: loaded (/etc/systemd/system/assist.service; enabled)
   Active: failed (Result: start-limit) since Sun 2018-09-02 06:44:18 PDT; 49s ago
  Process: 18286 ExecStart=/home/pi/AIY-projects-python/src/examples/voice/main.py (code=exited, status=1/FAILURE)
 Main PID: 18286 (code=exited, status=1/FAILURE)

Sep 02 06:44:18 raspberrypi systemd[1]: assist.service: main process exited,...E
Sep 02 06:44:18 raspberrypi systemd[1]: Unit assist.service entered failed s....
Sep 02 06:44:18 raspberrypi systemd[1]: assist.service holdoff time over, sc....
Sep 02 06:44:18 raspberrypi systemd[1]: Stopping Assist @ reboot...
Sep 02 06:44:18 raspberrypi systemd[1]: Starting Assist @ reboot...
Sep 02 06:44:18 raspberrypi systemd[1]: assist.service start request repeate....
Sep 02 06:44:18 raspberrypi systemd[1]: Failed to start Assist @ reboot.
Sep 02 06:44:18 raspberrypi systemd[1]: Unit assist.service entered failed s....
Hint: Some lines were ellipsized, use -l to show in full.

manoj7410, 2018-09-03:
@JITESHNAIR01 Did you follow all the instructions given at : https://aiyprojects.withgoogle.com/voice/#makers-guide-run-your-app-automatically

And are you able to start the assistant demos manually ?
Can you paste the full error ?

dmitriykovalev, 2018-11-17:
We have an example of `.service` file which should work:

https://github.com/google/aiyprojects-raspbian/blob/5c74ce5154fe9ecdab5901e01b567d47fbdc068f/src/examples/voice/assistant_grpc_demo.service#L1-L26

closed 2018-11-17 by dmitriykovalev