google/aiyprojects-raspbian #481: Board version compatibility with pi versions

genus1, 2018-09-05:
I have a pi3b+ with voice kit version 1.  I have looked everyplace to find what board works with which pi.  I find that nothing really works out of the box.  I flashed the chip and the desktop with all updates shows fine.  The wifi test says works ok.  The audio test says, "You have voice hat (V1) installed at index 0!  Did you hear the sound?"   I do not hear sound.  All is connected correctly, and I tested that the speaker is working with other hardware off the AIY board.

I am building a sophisticated robot using a pi3b+ for the heavy code stuff and arduino for sensors and drivers.   I need the pi3b+ for all the AI stuff.  I need to know if I pi3b+ works with version 1, or do I need a pi zero with version2 as a slave.

Does the version 1 work with pi3b+.  Does ANY version work with pi3B+?

manoj7410, 2018-09-05:
@genus1 I have a PI3B+ with voice V1 Hat and with sdcard image https://github.com/google/aiyprojects-raspbian/releases/tag/v20180803

And speaker is working fine out of the box. Which sdcard image do you have ?

dmitriykovalev, 2018-09-05:
@genus1 You should use the latest SD card image. You can also try to run `sudo dtoverlay googlevoicehat-soundcard` manually and see if that helps.

genus1, 2018-10-18:
I have tried about everything.  I even dropped from 64 gig to 32 gig to make sure formatted fat.  I can update and upgrade all the raspbian no problem.  Wifi test works, audio says, "You do not have any sound cards installed"  I used the dtoverlay above and another. I have version 2018-08-03.  I remember there was an 04 version but cannot find it.  I am going to build another raspberry pi3 with new voice hat to see if I get the same challenges.  I need version 1 because I want to easily access the servo and other hardware programming on the v1 board.

genus1, 2018-10-18:
I reconfigured and reboot and is says  you have voice hat (V1) installed at index 0!

But no sounds

Going to build another board as stated above.

genus1, 2018-10-18:
I ran arecord -l and aplay -l and they both detected the board and listed card and device as 0.  It sees the card but is is normal for card and device to be the same for play and record?

I also wrote the program in aiy .asoundrc and it was correct no typo but the system said command failed.

ALSA lib conf.c:1852(snd_config_load1) _toplevel_:18:0:Unexpected char
etc
Playback open error: -22, Invalid argument

genus1, 2018-10-22:
I built a new system from a new kit and it works through the voice check.  I believe the problem is possibly in the new release of raspbian.  I am going to build another image using raspbain and the hacks and see if it will work this this hardware.  If so then my original system hardware was bad.

genus1, 2018-11-09:
I finally got back to this project.  I now have three raspberry working.  It is NOT an issue of Pi3b or Pi3b+.  I have have the full system working with all demos on two boards.  I have determined either the original board or button is bad.  I will switch out the button tomorrow and see if it is the button or board.  I am receiving an error in _drivers/_button.py: GPIO.add_event_detect(self.channel, self.polarity)
RuntimeError: Failed to add edge detection.

genus1, 2018-11-09:
Not the button.  The software works in other boards.  Will need to swap the boards to see if it is the Pi or the sound hat.

dmitriykovalev, 2018-11-10:
@genus1 Regarding `add_event_detect`, please check this https://github.com/google/aiyprojects-raspbian/issues/516#issuecomment-435644762

genus1, 2018-11-10:
Let me recap.  I purchased on voice kit and it would not run at all.  I purchased three and all voice worked on pi3 and pi3b+, but only ONE would work all the way.  I purchased five total and got three to work.  The two failing had the error in _drivers/_button.py: GPIO.add_event_detect(self.channel, self.polarity  RuntimeError: Failed to add edge detection.

I exchanged two and purchased another four.  Only ONE worked, the other three had the Failed to add edge detection.  I then exchanged three for new boards and the first one failed with the add edge detection.  So out of 12 boards and testing 10 I returned here because more than 50% failure is a problem.

I noticed the post above to check #516.  I am sure it is not a pinout problem or none would work, but it could be in the software.  I am going to downgrade the RPi.GPIO to 0.6.3 and see if it fixes the issue. e.g. pip3 install RPi.GPIO==0.6.3 as recommended in #516.  I hope it does not wreck my other code.  Will let you know if it fixes the new failed board.

genus1, 2018-11-10:
Problem solved downgrading the RPi.GPIO to 0.6.3 fixes the issue. e.g. pip3 install RPi.GPIO==0.6.3 as recommended in #516.

Google needs to fix this issue as an update immediately.  I already returned seven kits that could have been fixed by this software issue.  I did notice the light does not cleanup properly when closing the app, which could be fixed along with the segmentation error when shutting down with control C.

avnishc18, 2018-11-30:
https://github.com/google/aiyprojects-raspbian/issues/481#issuecomment-430886265
how you solve this problem after this 

> I reconfigured and reboot and is says you have voice hat (V1) installed at index 0!
> 
> But no sounds
> 
> Going to build another board as stated above.

how you solve this problem after this ?

genus1, 2018-12-20:
Google has now updated the software.  I recommend a new install if you can. If not then downgrade your install for the RPi.GPIO as above.

closed 2018-12-20 by genus1