google/aiyprojects-raspbian #486: How to modify aiy audio driver to support 8KHz audio

titusece, 2018-09-10:
Hello,
I have 8KHz sampling frequency supported audio file and want to use with aiy stuff, but seems it supports default 16KHz, how can I modify the aiy for 8KHz ?
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/_drivers/_recorder.py

>     def __init__(self, input_device='default',
>                  channels=1, bytes_per_sample=2, sample_rate_hz=16000):

TO

>     def __init__(self, input_device='default',
>                  channels=1, bytes_per_sample=2, sample_rate_hz=8000):

Thanks for the support.

dmitriykovalev, 2018-09-10:
`_recorder.py` is a class to record audio from the microphone to the file. You already have a file, right? How are you going to use your file?

titusece, 2018-09-11:
Hi,
Thanks for your reply.
I am using the following code and it uses the aiyprojects-raspian code.
https://github.com/senyoltw/DANBOARD/blob/master/mod/detect_intent_stream.py
I have audio mic which supports 8KHz only.

dmitriykovalev, 2018-09-11:
Looks like you are not using VoiceKit then. `_recorder.py` just generates `arecord` command line and executes it. You can try to run generated command line directly from the console and see what happens and how it fails.

https://github.com/google/aiyprojects-raspbian/blob/93a6306438492464c25916b6b2f53d8e4750579b/src/aiy/_drivers/_recorder.py#L60-L68

titusece, 2018-09-12:
Thanks for the reply.
Modified this arecord stuff for my audio source and also modified the following file for 8KHz, then its working as expected.
https://github.com/senyoltw/DANBOARD/blob/master/mod/detect_intent_stream.py

>         sample_rate_hertz = 8000

Thanks again.

closed 2018-09-12 by titusece