google/aiyprojects-raspbian #488: Model 'object_detection' is not loaded.

RickZLLiao, 2018-09-13:
Hi 

   I retrain with MobileNet SSD V1 with 256x256 DM=0.125，but I got an issue as follw
Could anybody solve the issue ?
pi@raspberrypi:~/AIY-voice-kit-python/src/examples/vision $ ./object_detection.py --input glassskew_Resized.png --output glassskew_detected.png
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 128, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, self._tbuf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./object_detection.py", line 39, in main
    result = inference.run(image_center)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 100, in run
    return self._engine.image_inference(self._key, image, params)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 327, in image_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 191, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 143, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./object_detection.py", line 50, in <module>
    main()
  File "./object_detection.py", line 46, in main
    image.save(args.output)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 110, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 103, in close
    self._engine.unload_model(self._key)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 251, in unload_model
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 194, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'object_detection' is not loaded.

kissste, 2018-09-13:
I have had this issue before, something is wrong with your model.
The chip is not able to handle. Based on my web searching, nobody knows what is wrong. It simply fails during execution.

Solution - use a model that works to start with and retrain that one.

RickZLLiao, 2018-09-14:
Could you please provide the check point file & Mobile SSD V1 config file to me, then I could train it myself. thanks.

kissste, 2018-09-14:
You will find what you are looking for here:
https://github.com/google/aiyprojects-raspbian/issues/402
and
https://github.com/google/aiyprojects-raspbian/issues/314

RickZLLiao, 2018-09-14:
Today i retrain Embedded MobileNet V1 SSD in the tensorflow 1.10，compiled successful. But i got same problem - Model 'object_detection' is not loaded.
How to get rid of it ?

RickZLLiao, 2018-09-14:
my transfer command was as follow :
./bonnet_model_compiler.par --frozen_graph_path=./Glass_skewEmbeded_MobileNet_SSD_Res256_DM_P125/Glassskew_Embeded_Model_256_256_DM_P125_20180914_1857.pb --output_graph_path=./Glass_skewEmbeded_MobileNet_SSD_Res256_DM_P125/Glassskew_Embeded_Model_256_256_DM_P125_20180914_1857.binaryproto --input_tensor_name="Preprocessor/sub" --output_tensor_names="concat,concat_1" --input_tensor_size=256 --debug

weiranzhao, 2018-09-14:
@RickZLLiao @kissste 

VisionKit has very limited HW resources and it can only support particular architecture with specific configuration.

When you configure the training pipeline, start with embedded_ssd_mobilenet_v1_coco.config,
https://github.com/tensorflow/models/blob/master/research/object_detection/samples/configs/embedded_ssd_mobilenet_v1_coco.config
That one is VisionKit friendly.

We provided more details about your use case, search 
`How to train and deploy a customized object detection model` at 
https://aiyprojects.withgoogle.com/vision/#makers-guide

zhoujustin@ provided a pretrained checkpoint in issue 314

Let me know if this solves the issue or not. Thanks!

RickZLLiao, 2018-09-15:
@sainttelant
Thanks for your help! Would I have to reflesh Image first ?

RickZLLiao, 2018-09-17:
@sainttelant
I test 4 case，
(1)python3 test_run_model_on_bonnet.py --model_path /opt/aiy/models/mobilenet_ssd_256res_0.125_person_cat_dog.binaryproto --input_height 256 --input_width 256  ==>Default Model，It's ok.
(2)python3 test_run_model_on_bonnet.py --model_path /opt/aiy/models/Glassskew_Embeded_Model_256_256_DM_P125_20180914_1857.binaryproto ==> It's failed. (My training model - Embeded Mobile Model，256x256, depth Multiplier: 0.125 )，Error Messafe as follow...
(3)python3 test_run_model_on_bonnet.py --model_path /opt/aiy/models/pikachu_detector.binaryproto --input_height 256 --input_width 256 ==> It's failed. (.binaryproto was from https://github.com/giacomobartoli/vision-kit/tree/master/pikachu-detector)
(4)python3 test_run_model_on_bonnet.py --model_path /opt/aiy/models/cat_detector_cwh_180602.binaryproto --input_height 256 --input_width 256 ==>It's failed.(.binaryproto was from chadwallacehart's github ，https://github.com/chadwallacehart/aiy_custom_cat_detector)

Case (2)~(4)，has same error , as follow
Traceback (most recent call last):
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 128, in transact
    fcntl.ioctl(self._dev, SPICOMM_IOCTL_TRANSACT, self._tbuf)
OSError: [Errno 14] Bad address

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_run_model_on_bonnet.py", line 66, in main
    for i, result in enumerate(camera_inference.run()):
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 76, in run
    yield self._engine.camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 267, in camera_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 191, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 143, in transact
    raise SpicommInternalError
aiy._drivers._spicomm.SpicommInternalError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_run_model_on_bonnet.py", line 77, in main
    (i, fps, ','.join(output_tensor_str)))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 87, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 79, in close
    self._engine.stop_camera_inference()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 273, in stop_camera_inference
    self._communicate(request)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 194, in _communicate
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_run_model_on_bonnet.py", line 81, in <module>
    main()
  File "test_run_model_on_bonnet.py", line 77, in main
    (i, fps, ','.join(output_tensor_str)))
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board

RickZLLiao, 2018-09-17:
@sainttelant & @weiranzhao，I need your help.

weiranzhao, 2018-09-17:
@RickZLLiao Quick thing to check, are you on the latest SD card image?

RickZLLiao, 2018-09-17:
@weiranzhao，I'm not update to latestest SD card image. I'll try today, and let you know the result, thanks.

RickZLLiao, 2018-09-18:
@weiranzhao，After i update  the latestest SD card image (aiyprojects-2018-08-03.img)，i got errors as follow : (No such file or directory: '/dev/vision_spicomm')

 pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ python3 any_model_camera.py --model_path /opt/aiy/models/Glassskew_Embeded_Model_256_256_DM_P125_20180914_1857.binaryproto --input_height 256 --input_width 256
Traceback (most recent call last):
  File "any_model_camera.py", line 63, in <module>
    main()
  File "any_model_camera.py", line 56, in main
    with CameraInference(model) as inference:
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 94, in __init__
    self._engine = self._stack.enter_context(InferenceEngine())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 227, in __init__
    self._transport = make_transport()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 89, in make_transport
    return _SpiTransport()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 29, in __init__
    self._spicomm = _spicomm.Spicomm()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 289, in __init__
    self._dev = os.open(SPICOMM_DEV, os.O_RDWR)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/vision_spicomm'

RickZLLiao, 2018-09-18:
@weiranzhao，I try commands as follw, but i got same error 
==> sudo apt-get update; sudo apt-get upgrade; sudo reboot
==> dpkg -l aiy*
from : https://aiyprojects.withgoogle.com/help/#updating-your-sd-card

weiranzhao, 2018-09-18:
@RickZLLiao

That error seems like system issue. Could you try to reflash the sd card with latest image?

weiranzhao, 2018-09-18:
@dmitriykovalev 

Any idea on the error message after reflashing?

RickZLLiao, 2018-09-18:
@weiranzhao，After i update the latestest SD card image (aiyprojects-2018-08-03.img) & finishing update，I also found
message : ' Failed to start load kernel modules ' , when rebooting in the first line.

RickZLLiao, 2018-09-18:
@manoj7410，thanks your responding, i'll try it. And let you know the result.

RickZLLiao, 2018-09-19:
@weiranzhao 
@manoj7410 
@dmitriykovalev
After download the sdcard image from:  https://github.com/google/aiyprojects-raspbian/releases/tag/v20180803，i could excute my embedded_mobilenet_ssd model successfully。
I excute cat detection and pikachu detection also successfully.
Thanks all your help.