google/aiyprojects-raspbian #489: AIY voicehat sound stopped working

slloyd88, 2018-09-13:
Can you please help me debug why my voicehat stopped working?  It was working but now it is not. Not sure what I did that made it stop. I have two of them and it is not working on either anymore so I am sure it is a software issue.

pi@raspberrypi:~/AIY-projects-python/checkpoints $ ./check_audio.py
Playing a test sound...
Did you hear the test sound? (y/n) n
There may be a problem with your speaker. Check that it's connected
properly

Here is what I think you will ask for:

pi@raspberrypi:~/AIY-projects-python/checkpoints $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~/AIY-projects-python/checkpoints $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0

/boot/config.txt has the following:
dtoverlay=googlevoicehat-soundcard
dtoverlay=i2s-mmap

dmitriykovalev, 2018-09-13:
Can you please check everything with the latest SD card image from the [releases](https://github.com/google/aiyprojects-raspbian/releases) page?

dmitriykovalev, 2018-11-17:
Please reopen if that's still a problem.

closed 2018-11-17 by dmitriykovalev

phaniteja1, 2018-12-21:
I have the exact same problem. @slloyd88 Were you able to fix it?

slloyd88, 2018-12-21:
No. I finally gave up and did my own thing without the voicehat.  It turned
out to be a piece of junk.