google/aiyprojects-raspbian #490: CameraInference crashes consistently on Pi 3B (aiy.vision.inference.InferenceException)

chadwallacehart, 2018-09-13:
**Issue** 
Demos that use `CameraInference` crash a few minutes to a few hours after starting. I used the [face_detection_camera.py](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/face_detection_camera.py) example without parameters but this happens with all 'CameraInference' code I have used.

There is nothing obvious that instigates the crash. Sometimes it happens after a few hundred frames. Sometimes its many thousands.

**Error** 
```
Traceback (most recent call last):
  File "/usr/lib/python3.5/contextlib.py", line 345, in __exit__
    if cb(*exc_details):
  File "/usr/lib/python3.5/contextlib.py", line 291, in _exit_wrapper
    callback(*args, **kwds)
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 101, in <lambda>
    self._stack.callback(lambda: self._engine.stop_camera_inference())
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 315, in stop_camera_inference
    self._communicate_bytes(_REQ_STOP_CAMERA_INFERENCE)
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 246, in _communicate_bytes
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 70, in main
    (inference.count, inference.rate, len(faces), avg_joy_score(faces)))
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 134, in __exit__
    self.close()
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 128, in close
    self._stack.close()
  File "/usr/lib/python3.5/contextlib.py", line 313, in close
    self.__exit__(None, None, None)
  File "/usr/lib/python3.5/contextlib.py", line 360, in __exit__
    raise exc_details[1]
  File "/usr/lib/python3.5/contextlib.py", line 345, in __exit__
    if cb(*exc_details):
  File "/usr/lib/python3.5/contextlib.py", line 291, in _exit_wrapper
    callback(*args, **kwds)
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 98, in <lambda>
    self._stack.callback(lambda: self._engine.unload_model(key))
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 296, in unload_model
    unload_model=pb2.Request.UnloadModel(model_name=model_name)))
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 240, in _communicate
    return self._communicate_bytes(request.SerializeToString())
  File "/usr/local/lib/python3.5/dist-packages/aiy_projects_python-1.1-py3.5.egg/aiy/vision/inference.py", line 246, in _communicate_bytes
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'FaceDetection' is not loaded.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 76, in <module>
    main()
  File "face_detection_camera.py", line 72, in main
    camera.stop_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
```
**Environment**
Vision bonnet from a Vision Kit 1.1 installed on a Pi 3B.
Issue happened before and after reinstalling latest AIY Kit drivers.

**Attempts to fix**
I tried replacing the camera cable and replaced the camera itself with the same result.
I have not had issues with `ImageInference`, but have not tested that as much.
I have run other programs that use the PiCamera camera for long periods of time and they do not have issues (i.e. 'raspivid -p -f -t 0' overnight).

**Help!**
Could this be an issue with using the vision bonnet on the Pi 3? 
Any suggestions for how to debug this?
Could this just be a momentary miss in the camera sensor that could be ignored? Is there a way to catch that exception to test that?

dmitriykovalev, 2018-09-13:
@chadwallacehart Are you using the latest SD card image from [releases](https://github.com/google/aiyprojects-raspbian/releases) page? Could you please also share the output from `dmesg` command when everything crashes?

chadwallacehart, 2018-09-13:
There is a good chance I started with the `2018-08-03` image, but I am not 100% sure. Is there a way to check?

Note in my efforts to fix this, I tried upgrading to the latest version with:
`sudo apt-get update; sudo apt-get upgrade; sudo reboot`. That actually broke some elements (I got warnings of missing libraries when running the demos), 
so I followed the [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) instructions and did `sudo apt-get --reinstall install aiy-dkms leds-ktd202x pwm-soft` which got everything working again (expect the error in this issue).

Here is the `dmesg` output::
```
pi@aiypi3:~/AIY-projects-python/src/examples/vision $ dmesg | grep aiy
[    1.684350] systemd[1]: Set hostname to <aiypi3>.
[    3.968222] aiy-vision spi0.0: Initializing
[    3.971725] aiy-io-i2c 1-0051: Setting board type vision
[    3.972369] aiy-io-i2c 1-0051: Driver loaded
[    4.044685] aiy-vision spi0.0: Failed to bind reset GPIO
[    4.052465] aiy-vision spi0.0: Initializing
[    4.056312] aiy-vision spi0.0: Failed to bind reset GPIO
[    4.096954] pwm-aiy-io pwm-aiy-io: Driver loaded
[    4.097276] aiy-vision spi0.0: Initializing
[    4.105918] aiy-vision spi0.0: Failed to bind reset GPIO
[    4.121214] gpio-aiy-io gpio-aiy-io: Driver loaded
[    4.121792] aiy-vision spi0.0: Initializing
[    4.124219] aiy-adc aiy-adc: Vision bonnet ADC configuration.
[    4.128024] aiy-adc aiy-adc: Driver loaded
[    4.128145] aiy-vision spi0.0: Resetting myriad on probe
[    4.484968] aiy-vision spi0.0: Resetting myriad
[    6.631596] aiy-vision spi0.0: Writing myriad firmware
[   12.124466] aiy-vision spi0.0: Myriad booting
[   12.302544] aiy-vision spi0.0: Myriad ready
[  145.280521] aiy-vision spi0.0: Device opened: 0xb5e02400
[  163.403983] aiy-vision spi0.0: Incoming crc mismatch: slave f4760132 vs computed 5d78ebb5
[  216.487751] aiy-vision spi0.0: Incoming crc mismatch: slave d5d47657 vs computed 1d238348
[  244.488617] aiy-vision spi0.0: Incoming crc mismatch: slave 3cd3812c vs computed 1cb1b8e4
[  247.272642] aiy-vision spi0.0: Incoming crc mismatch: slave b3ba0ef8 vs computed b9e143c2
[  248.473633] aiy-vision spi0.0: Incoming crc mismatch: slave e1da8279 vs computed a227452e
[  254.007914] aiy-vision spi0.0: Incoming crc mismatch: slave 528eb4de vs computed fc15f937
[  257.271666] aiy-vision spi0.0: Incoming crc mismatch: slave 743859ae vs computed b6a69413
[  267.608348] aiy-vision spi0.0: Incoming crc mismatch: slave ac930b77 vs computed a5babe92
[  267.937622] aiy-vision spi0.0: Incoming crc mismatch: slave 32484f1c vs computed cbed98d2
[  270.536616] aiy-vision spi0.0: Incoming crc mismatch: slave 546c7b39 vs computed 1ff234de
[  276.613680] aiy-vision spi0.0: Incoming crc mismatch: slave 473c8f58 vs computed 15fee5f3
[  280.315488] aiy-vision spi0.0: Incoming crc mismatch: slave edb8a887 vs computed 2d02c42a
[  289.398402] aiy-vision spi0.0: Incoming crc mismatch: slave 8f497335 vs computed 75401eb2
[  294.472761] aiy-vision spi0.0: Incoming crc mismatch: slave 17e89a08 vs computed ff504e3b
[  296.861391] aiy-vision spi0.0: Incoming crc mismatch: slave 4ec8bca3 vs computed 3007f6e2
[  319.194259] aiy-vision spi0.0: Incoming crc mismatch: slave 484364b5 vs computed e346127c
[  339.810502] aiy-vision spi0.0: Incoming crc mismatch: slave 38bb2092 vs computed a8fbaf6c
[  367.789192] aiy-vision spi0.0: Incoming crc mismatch: slave 1c689428 vs computed 86f6a441
[  369.040110] aiy-vision spi0.0: Incoming crc mismatch: slave 67cb75db vs computed 5390a2b7
[  383.721965] aiy-vision spi0.0: Incoming crc mismatch: slave 9239d23e vs computed 591a5f87
[  393.788020] aiy-vision spi0.0: Incoming crc mismatch: slave 8a7c7099 vs computed 20f5389b
[  394.576893] aiy-vision spi0.0: Incoming crc mismatch: slave 4fe1a094 vs computed b3cc33b0
[  399.310006] aiy-vision spi0.0: Incoming crc mismatch: slave b871ecff vs computed d92c8fdc
[  404.257990] aiy-vision spi0.0: Incoming crc mismatch: slave 4060019c vs computed cce29e75
[  405.445986] aiy-vision spi0.0: Incoming crc mismatch: slave 439f8004 vs computed 1a2cbd13
[  407.781855] aiy-vision spi0.0: Incoming crc mismatch: slave 2aa3df67 vs computed b42b6a85
[  412.067948] aiy-vision spi0.0: Incoming crc mismatch: slave 9ab32fe4 vs computed f0c367c6
[  427.778879] aiy-vision spi0.0: Incoming crc mismatch: slave 0ccb7001 vs computed ddcace38
[  442.043789] aiy-vision spi0.0: Incoming crc mismatch: slave fa9f60c7 vs computed 4aaf2fbb
[  447.829679] aiy-vision spi0.0: Incoming crc mismatch: slave ed2cace3 vs computed 4d68a066
[  448.171859] aiy-vision spi0.0: Incoming crc mismatch: slave e8a8125a vs computed 5cd1a38f
[  450.885654] aiy-vision spi0.0: Incoming crc mismatch: slave ea001592 vs computed ae306336
[  454.379606] aiy-vision spi0.0: Incoming crc mismatch: slave 5aef9aae vs computed 5c52cea5
[  466.564675] aiy-vision spi0.0: Incoming crc mismatch: slave 41c4474c vs computed a411f941
[  485.983798] aiy-vision spi0.0: Incoming crc mismatch: slave ff501209 vs computed 01afd3b4
[  498.175513] aiy-vision spi0.0: Incoming crc mismatch: slave c2fb336c vs computed 65792c98
[  499.623563] aiy-vision spi0.0: Incoming crc mismatch: slave b03ecd0b vs computed 86dab5f4
[  508.749466] aiy-vision spi0.0: Incoming crc mismatch: slave e15f7de8 vs computed 693f90a0
[  543.960295] aiy-vision spi0.0: Incoming crc mismatch: slave 3aca9c89 vs computed 68458c1f
[... cut ...]
[43252.455641] aiy-vision spi0.0: Incoming crc mismatch: slave db9c3bd8 vs computed 992ea3b4
[43333.374704] aiy-vision spi0.0: Slave not ready after 1000 ms
[43333.374729] aiy-vision spi0.0: Failed to write header: -85
[43333.374744] aiy-vision spi0.0: Fatal error, resetting
[43333.389239] aiy-vision spi0.0: Resetting myriad
[43335.534584] aiy-vision spi0.0: Writing myriad firmware
[43340.291378] aiy-vision spi0.0: Myriad booting
[43340.469446] aiy-vision spi0.0: Myriad ready
[43340.530256] aiy-vision spi0.0: Device released: 0xb32c4e00
```

dmitriykovalev, 2018-09-14:
That's weird but something similar already happened before. Can you try solution from #418 by decreasing ARM frequency? Another option to consider:  https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/docs/vision.md#troubleshooting

chadwallacehart, 2018-09-14:
ok, tried both of those (independently, not at the same time) but unfortunately neither made a difference.

Remember this is a Pi 3, so I dropped the CPI from the usual 1.2GHz to 800MHz.

`dmesg | grep aiy` again for reference:
```
[    3.210143] systemd[1]: Set hostname to <aiypi3>.
[    5.389643] aiy-io-i2c 1-0051: Setting board type vision
[    5.399334] aiy-io-i2c 1-0051: Driver loaded
[    5.436080] pwm-aiy-io pwm-aiy-io: Driver loaded
[    5.457898] aiy-vision spi0.0: Initializing
[    5.469603] aiy-vision spi0.0: Failed to bind reset GPIO
[    5.481084] aiy-vision spi0.0: Initializing
[    5.482564] gpio-aiy-io gpio-aiy-io: Driver loaded
[    5.488166] aiy-vision spi0.0: Resetting myriad on probe
[    5.534740] aiy-adc aiy-adc: Vision bonnet ADC configuration.
[    5.540608] aiy-adc aiy-adc: Driver loaded
[    5.850185] aiy-vision spi0.0: Resetting myriad
[    7.991622] aiy-vision spi0.0: Writing myriad firmware
[   13.512338] aiy-vision spi0.0: Myriad booting
[   13.690406] aiy-vision spi0.0: Myriad ready
[  155.741012] aiy-vision spi0.0: Device opened: 0xb5b86600
[  444.787777] aiy-vision spi0.0: Slave not ready after 1000 ms
[  444.787788] aiy-vision spi0.0: Failed to write header: -85
[  444.787795] aiy-vision spi0.0: Fatal error, resetting
[  444.801406] aiy-vision spi0.0: Resetting myriad
[  446.947923] aiy-vision spi0.0: Writing myriad firmware
[  451.467957] aiy-vision spi0.0: Myriad booting
[  451.646016] aiy-vision spi0.0: Myriad ready
[  451.713756] aiy-vision spi0.0: Device released: 0xb5b86600
```

dmitriykovalev, 2018-09-17:
Did you try to use the same VisionBonnet with Pi Zero? Unfortunately, we didn't do much testing on Pi3 but it was working fine in all cases on our Pi3 boards. By the way, is camera working fine when you run `raspivid -t 0`?

chadwallacehart, 2018-09-17:
I ran `raspivid -p -f -t 0` overnight without issue.

I picked up a new Vision Kit to help try to isolate the issue, so I'll report back.

dmitriykovalev, 2018-09-19:
@chadwallacehart Please try to run old VisionKit on Pi Zero if possible.

chadwallacehart, 2018-09-23:
Ok, I did some additional tests. I was assuming that 
1. there is some incompatibility between the Vision Bonnet and the Pi 3B, or
2. My Vision Bonnet was malfunctioning or that 

First I test to see if the existing Vision Bonnet I was working by putting it on a the Pi Zero WH that came with the kit. I used a brand new SD card using the `aiyprojects-2018-08-03.img.xz` image. This resulted in the same error:
```
#14235 (12.30 fps): num_faces=0, avg_joy_score=0.00
#14236 ( 8.39 fps): num_faces=0, avg_joy_score=0.00
#14237 (10.77 fps): num_faces=0, avg_joy_score=0.00
Traceback (most recent call last):
  File "/usr/lib/python3.5/contextlib.py", line 345, in __exit__
    if cb(*exc_details):
  File "/usr/lib/python3.5/contextlib.py", line 291, in _exit_wrapper
    callback(*args, **kwds)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 101, in <lambda>
    self._stack.callback(lambda: self._engine.stop_camera_inference())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 315, in stop_camera_inference
    self._communicate_bytes(_REQ_STOP_CAMERA_INFERENCE)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 246, in _communicate_bytes
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: StopCameraInference: Camera inference is not running.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 70, in main
    (inference.count, inference.rate, len(faces), avg_joy_score(faces)))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 134, in __exit__
    self.close()
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 128, in close
    self._stack.close()
  File "/usr/lib/python3.5/contextlib.py", line 313, in close
    self.__exit__(None, None, None)
  File "/usr/lib/python3.5/contextlib.py", line 360, in __exit__
    raise exc_details[1]
  File "/usr/lib/python3.5/contextlib.py", line 345, in __exit__
    if cb(*exc_details):
  File "/usr/lib/python3.5/contextlib.py", line 291, in _exit_wrapper
    callback(*args, **kwds)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 98, in <lambda>
    self._stack.callback(lambda: self._engine.unload_model(key))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 296, in unload_model
    unload_model=pb2.Request.UnloadModel(model_name=model_name)))
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 240, in _communicate
    return self._communicate_bytes(request.SerializeToString())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 246, in _communicate_bytes
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'FaceDetection' is not loaded.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 76, in <module>
    main()
  File "face_detection_camera.py", line 72, in main
    camera.stop_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
```

Then I proceeded to test if my Vision Bonnet was faulty, so I removed the original Vision Bonnet and installed a new one that I just purchased. This also had the same result:
```
#47812 (11.24 fps): num_faces=0, avg_joy_score=0.00
#47813 ( 8.99 fps): num_faces=0, avg_joy_score=0.00
Traceback (most recent call last):
[... 34 lines cut ...]
    return self._communicate_bytes(request.SerializeToString())
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 246, in _communicate_bytes
    raise InferenceException(response.status.message)
aiy.vision.inference.InferenceException: UnloadModel: Model 'FaceDetection' is not loaded.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "face_detection_camera.py", line 76, in <module>
    main()
  File "face_detection_camera.py", line 72, in main
    camera.stop_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit__
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 752, in close
    raise exc
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1995, in _control_callback
    "No data recevied from sensor. Check all connections, "
picamera.exc.PiCameraRuntimeError: No data recevied from sensor. Check all connections, including the SUNNY chip on the camera board
```

I am not sure what else I can do to rule out hardware issues - I have now tried different camera cables, camera, vision bonnets, and Pi's. 
I suspect this is a software problem.

@dmitriykovalev - what do you recommend?

chadwallacehart, 2018-09-24:
I tried remaking the face_detection_camera.py code a bit to avoid using the inference.run generator and it seems to have fixed my issue:
```
from aiy.vision.inference import CameraInference
from aiy.vision.models import face_detection
from picamera import PiCamera
from time import time

def avg_joy_score(faces):
    if faces:
        return sum(face.joy_score for face in faces) / len(faces)
    return 0.0

with PiCamera(sensor_mode=4, resolution=(1640, 1232), framerate=30) as camera:
    camera.start_preview()

    with CameraInference(face_detection.model()) as inference:
        rate = 0.0
        count = 1
        now = None
        before = None

        while True:
            cam_state = inference.camera_state()
            now = time()
            rate = 1.0 / (now - before) if before else 0.0

            if cam_state.running:
                result = inference._engine.camera_inference()
                faces = face_detection.get_faces(result)

            else:
                print("Camera stopped")
                print(cam_state)
                break

            print('#%08d (%5.2f fps): num_faces=%d, avg_joy_score=%.2f' %
                (count, rate, len(faces), avg_joy_score(faces)))

            before = now
            count += 1
```

I ran this for 267120 frames (>7 hours) and it never stopped.

I then ran [face_detection_camera.py](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/vision/face_detection_camera.py) immediately after and that crashed with the error above in 39860 frames.

There isn't a while lot to the run function in the CameraInference class inside of [inference.py](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/aiy/vision/inference.py), but maybe there is some infrequent timing issue causing this?

What code do you use for your longevity tests?

chadwallacehart, 2018-09-25:
After some more tests my code above does still fail. In the same test it stopped after `22771` frames.  In a second it went for a long while at `504964` before the exception. I trying to see if I can catch and continue on the exception.

andrewtvuong, 2018-10-12:
I face the same problem. Though I suspect the Pi is also partially to blame. Even without the Google Kit it occasionally slips, so I've written a restart script. I have managed to get it to livestream for over a month with about 10 crashes that were handled.

rnehrboss, 2019-02-06:
@andrewtvuong ,  Hey Andrew... do you have that script you'd be willing to share?
BTW I find your blog/med school change of path interesting.
Thanks!
Richard

andrewtvuong, 2019-02-17:
@rnehrboss thanks! this has been a while ago, which I've abandoned. Pi is too unreliable due to the SD card. They turn out to be corrupted or I get frustrated enough that I keep reflashing the image to the point that the SD card fails. I am waiting for the Edge TPU, which has onboard storage.

essentially I used ffmpeg input video output rtsp stream. make sure to include audio as some servers will reject video without audio. you can do something called audio copy, basically feed it internal audio so no microphone or anything required. I think the most recent AIY image broke audio from many of my programs, so I am now pursuing other things.

WP-tim, 2019-03-03:
I found that using "preview" through The Vision Bonnet APIs was unstable (issues as described above).
If I instead display through [PyGame ](https://www.raspberrypi.org/forums/viewtopic.php?t=181301)the Kit is very stable (over 30 days uptime) but the image has significant lag
`sudo apt-get install python3-pygame`

WP-tim, 2019-03-10:
I used a non-AIY script based on [pygame ](https://www.raspberrypi.org/forums/viewtopic.php?t=181301).
I left the hardware intact; the script simply doesn't import any AIY modules.
It too ran 30+ days.

HOWEVER, I added `import pygame` into the AIY script, and again the system is unstable. Errors (as above) within 1 hour!
During testing the camera is covered, so very boring and consistent video.

@dmitriykovalev throughout the threads I have **not** seen this correlation mentioned.  When you've tested, do you enable camera.start_preview?  would you consider testing with `import pygame` in your list of modules?
More importantly, Dmitri, **are there still activities to correct/improve** the AIY-Vision kits?

(note, my severely hacked, and not-cleaned, script attached)
[AIYframe.py.txt](https://github.com/google/aiyprojects-raspbian/files/2949966/AIYframe.py.txt)