google/aiyprojects-raspbian #492: How to use AIY in the official system of raspberry?

zouzhe1, 2018-09-15:
I didn't use the img provided by the AIY project, so how should my sound driver, speaker and microphone driver be set? Is ALSA using?

my AIY is  voice v1

zouzhe1, 2018-09-15:
And Is this aiy project hardware open source? Where can I find the SCH and PCB for voice v1?

dmitriykovalev, 2018-09-17:
@zouzhe1 Regarding you second question: SCH and PCB are not available for public. Regarding the first question, please take a look at https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/docs/voice.md#drivers. You need to enable VoiceHAT overlay on the Raspbian image and everything should start working. You probably also need to comment out `dtparam=audio=on` in the `/boot/config.txt` to make VoiceHAT a single sound device.

mpember, 2018-10-26:
> And Is this aiy project hardware open source? Where can I find the SCH and PCB for voice v1?

There are some who have created alternative HATs that provide near-identical functionality.
https://github.com/shivasiddharth/custom-voice-hat

dmitriykovalev, 2018-11-17:
We've added VoiceHAT schematic files to the `docs` folder: https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/docs/design/voice_hat

closed 2018-11-17 by dmitriykovalev