google/aiyprojects-raspbian #495: After training with Embedded mobilenet SSD, but detection not thing

RickZLLiao, 2018-09-21:
Were anybody training with embedded mobilenet ssd, but detecting not thing.
I train with pretrain file from  zhoujustin. ==>https://drive.google.com/file/d/1_MeZ8kvmpNibPZvSJGnwKNRATeuyxNtu/view
Then zhoujustin saved the day and shared a trained model based on the 20-class VOC dataset.
I had trined 30000 epoch. But i couldn't detect my training images.
If anyone could provide me some suggestions ? Thanks for your swift reply.

RickZLLiao, 2018-09-21:
My learning rate was e-4. But detect nothing, so i change to e-6. After training 300000 epoch, it couldn't detect my target.

RickZLLiao, 2018-09-25:
I had predicted embedded mobilenet ssd v1 in the PC, and i could detected my target。When i loaded model in the vision kit, it couldn't detect angything. I had lower my min threshold, but i got nothing.
What's wrong with my setting ?

RickZLLiao, 2018-09-28:
I check the score, and i found background scores were larger than target scores.  I try to label larger target , and train it. I found target scores could larger than background scores. And i could detect target object.
Why i must label larger target image, then i could get larger scores ?

zye1996, 2018-12-04:
Your training rate is too small. Change back to 0.004 would make a difference.

soufianesabiri, 2019-03-17:
I trained MobileNet SSD 300 for 3 days and still no detection... Please help!