google/aiyprojects-raspbian #500: _player.py play_wave() not working

pierrextardif, 2018-10-04:
I got an error when trying to play a wav file.

I believe the error comes from the with statement in the _player.py : 
```         with wave.open(wav_path, 'r') as wav: ```

Replaced it with :
```         wav = wave.open(wav_path, 'r') ```

and it worked.

manoj7410, 2018-10-04:
@pierrextardif Did you make any changes in Python version  ?

pierrextardif, 2018-10-04:
No, 

I use python 3.

line 64, I changed the with statement:
```with wave.open(wav_path, 'r') as wav:```

Replaced it with :
```wav = wave.open(wav_path, 'r')```

jfatula, 2018-10-04:
Looks like context manager support using with was added for wave.open() at Python 3.4. 
[https://python.readthedocs.io/en/latest/whatsnew/3.4.html](url)

dmitriykovalev, 2018-10-05:
Please update you python version, @jfatula is right.

closed 2018-10-05 by dmitriykovalev