google/aiyprojects-raspbian #503: Assistant_library_demo doesn't detect the hotword"ok google", latest image, aiy voice kit v1 on rpy 3b

blandiant, 2018-10-12:
Hi
I use the latest image, on a rpy 3B with aiy voice kit v1
As said in the title, with assistant_library_demo, the assistant doesn't detect the hotword "Ok google" (it work only 1 time out of 20,
while the google assistant on my phone detect it each time)
Except that everything else works just fine, i checked the volume of the mic, it is good(I turned up the volume to 200%: see issue #356) , When i use the assistant_grpc_demo  it understand what i say just fine 
When i am lucky and the assistant_library_demo detect the "Ok google" it understand what i say after too

mpember, 2018-10-17:
@blandiant , is it possible that your microphone has fallen slightly out of place?  It is possible that the sensitivity of the hotword detection is such that it expects a slightly clearer audio than the regular voice recognition.

blandiant, 2018-10-20:
Hi,
Sorry for this late answer,
you were right, the accent is the problem, i speak and use the aiy voice kit in french, when i speak with the north American accent, it work.
 I guess it is not possible to change the accent "detection"?
Thank you

closed 2019-11-14 by dmitriykovalev