google/aiyprojects-raspbian #505: Update kills vision

WP-tim, 2018-10-14:
Using prompted update kills the vision bonnet.

Using the v1.1 AIY Vision kit (purchased Aug 3 2018)
Reimage SD card with [aiyprojects-2018-08-03.img.xz](https://github.com/google/aiyprojects-raspbian/releases)
At this point the bonnet works (privacy LED lights, tone sounds, button colors with a smile).
the OS version (uname -a) is: Linux raspberrypi 4.14.52+ #1123 Wed Jun 27 17:05:32 BST 2018 armv6l GNU/Linux

HOWEVER
The desktop window asks to change the password, then prompts to update the OS.
after _that_ update, the bonnet **_no longer works_**.
version: Linux raspberrypi 4.14.71+ #1145 Fri Sep 21 15:06:38 BST 2018 armv6l GNU/Linux

pi@raspberrypi:~ $ sudo systemctl status  joy_detection_demo
```
● joy_detection_demo.service - AIY Joy Detection Demo
   Loaded: loaded (/lib/systemd/system/joy_detection_demo.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-10-14 07:55:59 PDT; 1min 42s ago
 Main PID: 229 (code=exited, status=1/FAILURE)

Oct 14 07:54:56 raspberrypi systemd[1]: Started AIY Joy Detection Demo.
Oct 14 07:55:57 raspberrypi python3[229]: ERROR:__main__:AIY Vision Bonnet is not attached or not configured properly.
Oct 14 07:55:59 raspberrypi systemd[1]: joy_detection_demo.service: Main process exited, code=exited, status=1/FAILURE
Oct 14 07:55:59 raspberrypi systemd[1]: joy_detection_demo.service: Unit entered failed state.
Oct 14 07:55:59 raspberrypi systemd[1]: joy_detection_demo.service: Failed with result 'exit-code'.
```

lsmod | grep aiy **returns nothing**
lsmod | grep pwm_soft **returns nothing**

Reverting to the 2018-08-03 image restores functionality.
Please **_do_** provide instructions to upgrade properly.
Please **_don't_** prompt to upgrade if that breaks the system.

manoj7410, 2018-10-16:
@WP-tim apt-get update and upgrade is breaking the AIY modules and is a known issue as per AIY Tech Support response. So, an upgrade needs to be avoided.

asmagill, 2018-10-17:
@manoj7410, do you know if they are working on fixing it or is that the answer, "Never update the OS"?

dmitriykovalev, 2018-10-17:
We are working on the fix in the next SD card image release.