google/aiyprojects-raspbian #513: AIY Vision not working

trsbalaji, 2018-10-26:
Hi, I bought vision kit version 1.1. Normal Raspberry pi zero w boots fine. Once connected MIPI flat flex cable to Raspberrypi zero w board is not booting.

https://aiyprojects.withgoogle.com/vision-v1/

Please help me.
[image]
[image]
[image]

dmitriykovalev, 2018-10-26:
Did you lock a small latch on the VisionBonnet's flex connector? Please also attach photograph of the bonnet board where flex contacts are visible.

trsbalaji, 2018-10-27:
Hi dmitriy, Thanks for your response. Connected flex cable board is not booting. Attached: photos

Could you please help me.

Thanks & Regards,

Balaji Babu

Attached: 
[image]
[image]
[image]

dmitriykovalev, 2018-10-27:
VisionBonnet on your photo is not connected correctly. You connected short flex to the camera connector and that's wrong. Please carefully follow the steps from https://aiyprojects.withgoogle.com/vision/#assembly-guide

trsbalaji, 2018-10-29:
Hi Dmitriy,

Thanks for your suggestion. Normal raspberry pi zero works fine after connecting vision bonnet getting below error.

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

Could you please help me.

Regards

Balaji Babu

manoj7410, 2018-10-29:
@trsbalaji Can you make sure that you have stopped joy_detection_demo before trying another demo ?

Command to stop joy_detection is : "sudo systemctl stop joy_detection_demo"

trsbalaji, 2018-10-30:
> @trsbalaji Can you make sure that you have stopped joy_detection_demo before trying another demo ?
> 
> Command to stop joy_detection is : "sudo systemctl stop joy_detection_demo"

Thanks for your timely help. Could you please tell me, where i will find to download  cat.jpg, dog.jpg file

trsbalaji, 2018-10-30:
> VisionBonnet on your photo is not connected correctly. You connected short flex to the camera connector and that's wrong. Please carefully follow the steps from https://aiyprojects.withgoogle.com/vision/#assembly-guide

Finally worked.  Thanks for your timely help

Could you please tell me, where i will find to download  cat.jpg, dog.jpg file

Thanks,
Balaji Babu

dmitriykovalev, 2018-10-30:
Test images can be downloaded using `make` command: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/tests/images/Makefile

trsbalaji, 2018-10-31:
> Test images can be downloaded using `make` command: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/tests/images/Makefile

Hi, Thanks for your help. After download the hotdog.jpg image i run the program getting error message.

Sorry for the disturb. Could you please help me

pi@raspberrypi:~/AIY-projects-python/src/examples/vision $ object_detection.py  --input=hotdog.jpg --output=/home/pi/AIY-projects-python/src/examples/vision/result.jpg
Traceback (most recent call last):
  File "/usr/local/bin/object_detection.py", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/opt/aiy/projects-python/src/examples/vision/object_detection.py", line 60, in <module>
    main()
  File "/opt/aiy/projects-python/src/examples/vision/object_detection.py", line 50, in main
    result = inference.run(image_center)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 98, in run
    return self._engine.image_inference(self._key, image, params)
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 317, in image_inference
    return self._communicate(request).inference_result
  File "/opt/aiy/projects-python/src/aiy/vision/inference.py", line 187, in _communicate
    response.ParseFromString(self._transport.send(request.SerializeToString()))
  File "/opt/aiy/projects-python/src/aiy/_drivers/_transport.py", line 33, in send
    return self._spicomm.transact(request)
  File "/opt/aiy/projects-python/src/aiy/_drivers/_spicomm.py", line 117, in transact
    raise SpicommOverflowError(PAYLOAD_SIZE)
aiy._drivers._spicomm.SpicommOverflowError

dmitriykovalev, 2018-10-31:
Please download and flash the latest SD card image from https://github.com/google/aiyprojects-raspbian/releases

closed 2018-11-16 by dmitriykovalev