google/aiyprojects-raspbian #515: Audio does not work after re-imaging/installing voice kit 2 Pi Zero (Original HW)

tibre, 2018-10-28:
I have spent a few hours chasing this one without luck, so decided to ask for some guidance...

**Problem**
'Do not have sound card error message' - after reimaging the SD card and installing with the latest build available.

- I am using the original hardware.  AIY voice kit v2.0
- BUILD: [(2018-08-03)](https://github.com/google/aiyprojects-raspbian/releases/tag/v20180803).
- Tried adding a line to the config.txt file 'dtoverlay' below.  But it should not apply in my case - either way it did not work

**Error/Details**
- When running the check audio script I get the following error message (I have a voice bonnet so I don't think the suggestion below applies):

_You do not have any sound cards installed. Please check that AIY sound card is
properly connected.
For some Voice HATs (not Voice Bonnets!) you need to add the following line
to /boot/config.txt:
dtoverlay=googlevoicehat-soundcard
To do that simply run from a separate terminal:
echo "dtoverlay=googlevoicehat-soundcard" | sudo tee -a /boot/config.txt
Press Enter to close..._

**Any help will be appreciated!**

manoj7410, 2018-10-31:
@tibre Can you paste the complete error ? And can you also share the output of the command "lsmod".

tibre, 2018-11-04:
I made it work in a couple of ways:

1. I installed the 2018-04-13 build and it worked without issues.  
2. Installed a fresh copy of raspbian (NOOBS v2_9_0) and followed the HACKING.md instructions.  There was a problem with the sound card driver enumeration Index (bcm2835 ALSA was enumerated first, Voicebonnet second).  Commented out '#dtparam=audio=on' on /boot/config.txt and that solved it (Dimitry's suggestion)

**So** I don't know why 2018-08-03 does not work with the kit right off the bat, but thanks for following up.

tibre, 2018-11-04:
> @tibre Can you paste the complete error ? And can you also share the output of the command "lsmod".

Well, with #2 above (Fresh copy of rasbian and following HACKING) I am able to test speaker and record but when trying the assistant_grpc_demo I get an error message:

Failed to import google.cloud.speech. Try:
    env/bin/pip install -r requirements.txt

I am not running a virtual environment.  Also, just replying to your previous ask here the lsmod output of my current configuration:
_--------
Module                  Size  Used by
fuse                   92909  3
rfcomm                 37969  4
bnep                   11981  2
hci_uart               27218  1
btbcm                   8248  1 hci_uart
serdev                  9370  1 hci_uart
bluetooth             366847  29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           17649  1 bluetooth
aiy_adc                 4690  0
gpio_aiy_io             4226  0
evdev                  11644  1
industrialio           63573  1 aiy_adc
pwm_aiy_io              5084  0
leds_ktd202x            7349  0
aiy_io_i2c              6848  3 aiy_adc,gpio_aiy_io,pwm_aiy_io
brcmfmac              273230  0
brcmutil                9114  1 brcmfmac
cfg80211              542242  1 brcmfmac
rfkill                 21476  6 bluetooth,cfg80211
snd_aiy_voicebonnet     5048  3
rt5645                 88615  2 snd_aiy_voicebonnet
rl6231                  2904  1 rt5645
snd_soc_bcm2835_i2s     7385  2
regmap_mmio             3874  1 snd_soc_bcm2835_i2s
snd_soc_core          178253  3 snd_aiy_voicebonnet,rt5645,snd_soc_bcm2835_i2s
snd_compress            9956  1 snd_soc_core
snd_pcm_dmaengine       5370  1 snd_soc_core
i2c_bcm2835             6465  0
snd_pcm                89558  4 snd_pcm_dmaengine,rt5645,snd_soc_bcm2835_i2s,snd_soc_core
snd_timer              22428  1 snd_pcm
snd                    60154  10 snd_compress,snd_timer,snd_soc_core,snd_pcm
uio_pdrv_genirq         3718  0
uio                     9901  1 uio_pdrv_genirq
fixed                   3033  0
pwm_soft               25457  0
i2c_dev                 6674  0
ip_tables              12427  0
x_tables               22130  1 ip_tables
ipv6                  397278  30_

dmitriykovalev, 2018-11-04:
@tibre Can you please confirm that you used 2018-08-03 image with the VisionBonnet (v2.0) and sound check failed with `You do not have any sound cards installed` message right after the boot? It's pretty weird, especially if the same sound check works fine with 2018-04-13 build. 

Can you please also attach the output of `dmesg` command right after the check fails?

tibre, 2018-11-04:
@dmitriykovalev Yes, that the file:

**MD5**
MD5 hash of aiyprojects-2018-08-03.img.xz:
386dafcc76c8539990e75e374f33b513
CertUtil: -hashfile command completed successfully.

**VisionBonnet (v2.0)**

I will reimage an SD with that build and will give you details.  Now, the only difference that I recall and that might be creating the problem is updates.  When installing 08-03, updates are an option and they are executed right away compared to 04-13.

dmitriykovalev, 2018-11-04:
@tibre Sorry, right, that was a typo, VoiceBonnet :)  Please do not install any updates for now (we are working to fix this) That could be exactly the source of problems.

tibre, 2018-11-04:
@dmitriykovalev Thanks for the update.  
That said, any ideas.suggestions how to move forward pass this message (when trying latest Noobe's install+HACKING instructions):

_Failed to import google.cloud.speech. Try:
env/bin/pip install -r requirements.txt_

dmitriykovalev, 2018-11-12:
@tibre That's weird, how do you run cloud speech example, what's the command line?  This error message is obsolete but still shouldn't happen. Are you using `python3`?

dmitriykovalev, 2018-11-12:
@tibre Have you installed https://pypi.org/project/google-cloud-speech/ library?

```bash
sudo pip3 install google-cloud-speech
```

andrewtvuong, 2018-11-29:
I've noticed certain programs no longer produce sound from 4-13-18 to 8-3-18. I'm not certain but it seems the culprit has something to do with switching to PulseAudio from ALSA? I don't know much about the subject.

andrewtvuong, 2018-11-29:
> @tibre Have you installed https://pypi.org/project/google-cloud-speech/ library?
> 
> ```shell
> sudo pip3 install google-cloud-speech
> ```

wow that's awesome! I'm new to the subject, but is there a way to have its model so I can run on device?

dmitriykovalev, 2018-11-30:
@andrewtvuong Yes, we switched to PulseAudio (it still uses ALSA underneath). Which programs don't work for you?

mdbrewer, 2018-12-28:
I am having the same issue with two out of the box AIY kits, using the 2018-11-16 build. 

You do not have any sound cards installed. Please check that AIY sound card is
properly connected.

For some Voice HATs (not Voice Bonnets!) you need to add the following line
to /boot/config.txt:

dtoverlay=googlevoicehat-soundcard

To do that simply run from a separate terminal:

echo "dtoverlay=googlevoicehat-soundcard" | sudo tee -a /boot/config.txt
Press Enter to close...

I have updated the firmware, APT-upgrade, and checked the help forum. Still no go.

mdbrewer, 2018-12-28:
I loaded AIY Kits Release 2018-04-13 and things work. I am not sure the difference.

TimeDoctor, 2019-01-20:
this issue has been going on since November? I also have the issue with "You do not have any sound cards installed. "

TimeDoctor, 2019-01-20:
I have this "You do not have any sound cards installed." issue as well with the check_audio script on the v2 pi zero kit

mongonta0716, 2019-01-25:
Hi,

I think that the cause is that the driver disappears by updating firmware.
Try the following commands.

cd /usr/src
sudo dkms install aiy-1.1
sudo dkms install aiy-vision-1.1
sudo dkms install aiy-voicebonnet-soundcard-1.0
sudo dkms install leds-ktd202x-1.1
sudo dkms install pwm-soft-1.1
sudo reboot

The name of the driver changes depending on the AIY Kits OS version, check with 'ls /usr/src'

TimeDoctor, 2019-01-25:
> sudo dkms install aiy-voicebonnet-soundcard-1.0

thanks, this one did it for my project, once I made it 1.1 sound worked again. Oddly, all the other modules that were useful for the voice kit were already installed!

andrewtvuong, 2019-01-25:
> Hi,
>
> I think that the cause is that the driver disappears by updating firmware.
> Try the following commands.
>
> cd /usr/src
> sudo dkms install aiy-1.1
> sudo dkms install aiy-vision-1.1
> sudo dkms install [...]

Hi, thanks for sharing.

```
Error! DKMS tree already contains: aiy-voicebonnet-soundcard-1.0
You cannot add the same module/version combo more than once.
```
I'm getting this message for every command you listed. Should I uninstall and reinstall? If so I'm not familiar and don't want to mess up dependencies or some kind of problem, would it be safe to uninstall and how? rm -rf?

mongonta0716, 2019-01-26:
Hi @andrewtvuong 

Sorry I couldn't help you. 
If there is a driver it is a different case from me.I have written something unrelated to this Issue.In my case,the following command was executed and the driver was built.
``
sudo dkms install aiy-voicebonnet-soundcard-1.0
``

My environment
Google AIY Voice Kit V2
BUILD: 2018-11-16 and apt-upgrade
firmware:4.14.79+

mdbrewer, 2019-03-10:
> Hi,
>
> I think that the cause is that the driver disappears by updating firmware.
> Try the following commands.
>
> cd /usr/src
> sudo dkms install aiy-1.1
> sudo dkms install aiy-vision-1.1
> sudo dkms install [...]

After a fresh install of 11-16 image, an update, then running the above driver installs, and it works.

remzioten, 2019-03-21:
> Hi,
>
> I think that the cause is that the driver disappears by updating firmware.
> Try the following commands.
>
> cd /usr/src
> sudo dkms install aiy-1.1
> sudo dkms install aiy-vision-1.1
> sudo dkms install [...]

This solved for me. Thanks

jreyn88, 2019-05-10:
Why would an image created by starting with AIY Voice then adding Shairport-Sync and Homebridge with V2 voicebonnet, work in another 3b but not the Zero WH that came with the voicebonnet? I tried the above steps but it recognizes a Google soundcard in the Zero instead of the voicebonnet in both 3b's.

andrewtvuong, 2019-05-10:
Very strange, but I had a lot of problems on 3A+, that resolved itself after several reimages. Keep downloading and changing SD cards, update etcher. I don’t understand the reasons, but all my problems are gone.

juanfonsecasolis, 2019-06-23:
I downloaded the 2018-08-03 build and ended up repeating timbre's experience. Could you remove that build to other people stop experiencing problems, please?

Trixter313, 2020-06-20:
I am having this same issue on the 2019-11-13 build, should I just flash the older version?

rudjer, 2020-09-30:
I'm having this same issue on a V2 voice bonnet with the 2019-11-13 build running:  
~/AIY-projects-python/checkpoints $ python3 check_audio.py
I tried adding to /boot/config.txt and rebooting, but that didn't help, perhaps not
surprisingly given that the message only implies this might help for V1.

diegoshh, 2020-10-21:
Hi, I have a problem with audio, after intalling aiy-2019-11-13.img.xz on the raspberrypi the icon for the speaker on the navigation bar dissaperas, also I have that same message (you so not have any sound card installed. Please check that AIY sound card is properly connected.

I wanted to use the speakers of a samsung TV, but it didnt work, so I connected a speaker in the AUX and it still doesnt work when I check the audio. Help please!

diegoshh, 2020-10-21:
by the way Im using a raspberry pi 4