google/aiyprojects-raspbian #516: Voice V1 button.wait_for_press() failed to add edge detection

choconnit, 2018-10-29:
Hello,
I tried the example cloudspeech_demo.py, and get_led() returned a warning: “channel is already in used" then segmentation faulted on button.wait_for_press with runtime error: failed to add edge detection.  
Please help, my Voice Kit version 1 is running the 2018-08-03 image on a RPi 3B+. All the check routines work fine. 
Thank you,
DT

dirkk0, 2018-10-30:
same here.
could it be that the actual version is wired for the zero, and the pins changed so the V1 won't work?

choconnit, 2018-11-02:
> same here.
> could it be that the actual version is wired for the zero, and the pins changed so the V1 won't work?

I'm sure there's a workaround somewhere!

9600-, 2018-11-02:
Just wanted to add to this issue, as I am having problems with any example script referencing the GPIO pins for the button. Using Kit V1 with a Raspberry Pi 3B+. 

```
src/examples/voice/assistant_grpc_demo.py/opt/aiy/projects-python/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
[2018-11-02 14:43:34,320] INFO:recorder:started recording
Press the button and speak
Traceback (most recent call last):
  File "src/examples/voice/assistant_grpc_demo.py", line 54, in <module>
    main()
  File "src/examples/voice/assistant_grpc_demo.py", line 39, in main
    button.wait_for_press()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_button.py", line 64, in wait_for_press
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/opt/aiy/projects-python/src/aiy/_drivers/_recorder.py", line 105, in run
    self._arecord.kill()
AttributeError: 'NoneType' object has no attribute 'kill'

    GPIO.add_event_detect(self.channel, self.polarity)
RuntimeError: Failed to add edge detection

```

9600-, 2018-11-02:
My guess is this is possible GPIO pinout changes, or edge detection parameter differences, between the V1 Kit's intended platform (RPI 2B/3B) and the RPI 3B+.

If we could get some verification on this, it would be much appreciated.
These kits will be used in an educational lab, and the combination of 3B+ and V1 kits are all we have.

choconnit, 2018-11-03:
After some trials and errors on Bluetooth pairing with Xbox One and my RPi 3B+, Google voice examples worked flawlessly.  I can’t recall the steps but will try to retrace the history.  

Anyhow, when I boot my RPi 3B+ with a fresh image from 2018-08-03, Voice V1 works.  Then, right after I did update and upgrade, the examples stopped working with the same edge detection error.

dirkk0, 2018-11-03:
I can confirm that.
If you skip the update and upgrade, and just install the json files, the button works.
Once you 'apt-get update' and -upgrade and reboot, the error is there.

lurfMalderFulder, 2018-11-03:
Downgrading RPi.GPIO to 0.6.3 fixes the issue. e.g. pip3 install RPi.GPIO==0.6.3

dmitriykovalev, 2018-11-04:
@9600- I actually saw the same `RuntimeError: Failed to add edge detection` issue recently. As @lurfMalderFulder already noticed, old version of `RPi.GPIO` doesn't have this problem.

dirkk0, 2018-11-04:
yes, that was it - thank you!

dirkk0, 2018-11-04:
... and maybe unrelated, but I had to edit ```/etc/pulse/default.pa``` to get it to work.
Strangely enough: when I started the python script directly, everything worked fine. One I started it in a screen session on startup, it didn't. The error was
```ERROR:mic_failure_detector.cc(140)] [assistant] Channel 1 has failed``` and editing this file helped.

I needed to uncomment line 38 and 39 in ```/etc/pulse/default.pa``` :
```
load-module module-alsa-sink
load-module module-alsa-source device=hw:1,0
```

dmitriykovalev, 2018-11-04:
@dirkk0 Please create another issue with full description to track the problem with `/etc/pulse/default.pa`.

choconnit, 2018-11-16:
Voice V1 works with the latest image 11-16-18.  That's after update/upgrade, without RPi.GPIO downgrade.  Thank you :)

dmitriykovalev, 2018-11-17:
We released [new SD card image](https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/docs/design/voice_hat) which has updated implementation of `wait_for_press`:
https://github.com/google/aiyprojects-raspbian/blob/5c74ce5154fe9ecdab5901e01b567d47fbdc068f/src/aiy/board.py#L101-L104

closed 2018-11-17 by dmitriykovalev