google/aiyprojects-raspbian #518: output node named "<output node>" does not appear in the graph

llealgt, 2018-10-31:
1
down vote
favorite
I'm trying to deploy to a google aiy vision kit v2 a custom model trained with Keras API in Tensorflow,when compiling the model from a .pb Tensorflow file with the vision kit compiler i get:

output node named "<output node>" does not appear in the graph.
My question is, how can i get from tensorboard the output node passed to the compiler in the parameter --output_tensor_names?

I already tried the values returned by:

keras_model.outputs[0].op.name    # returns final_output/Sigmoid
keras_model.outputs[0].name    # returns final_output/Sigmoid:0
and final_output, which is the name given to the last layer in keras.

But none of these worked, how can i see the correct expected node in tensorboard?

weiranzhao, 2018-11-05:
@llealgt May I know what model architecture are you using? 

Please note that due to limited resource on the device, only particular model architecture (with specific configuration) can run on the device. See here (https://aiyprojects.withgoogle.com/vision/#makers-guide--tensorflow-model-compiler)

Usually, the value to feed is the name of your final output tensor.

llealgt, 2018-11-05:
Thanks for your answer and help, yes i'm aware of the limitations in architecture , but in this case i'm using mobilenetV1 which the documentation says it's supported ,i also tried to write the .pb file as text to check if keras was doing some kind of renaming of tensors (which i confirmed was not done)  and i confirmed the output tensor is named "final_output/Sigmoid" but still the compiler cannot find it. Can i know how the compiler looks for the tensor passed as argument? maybe that would give me a hint

weiranzhao, 2018-11-05:
@llealgt The compiler will use the output tensor name information to find a set of nodes that must have for the inference to run. If the output tensor name is wrong, it won't be able to figure out the set of nodes to do inference. 

Do you mind to share a version of your pb file? That can help debugging.

llealgt, 2018-11-06:
I was not able to attach it here(it exceeds the 10 mb limit), so i pushed to one of my repos: https://github.com/llealgt/AIY_experiments/blob/master/trained_models/eagle_eye3.pb

weiranzhao, 2018-11-07:
@llealgt 
I took a closer look at your file. It seems to me this graph is a training graph, not a inference graph. Training graph contains a lot more operators (e.g., conditional branching) that our on device inference engine can not handle. Could you try to export a inference graph and give it another shot?

In addition, I'd suggest to try to train your model with the good old slim api. I know it is being deprecated. But it is the most reliable way to train a MobileNet model. 
https://github.com/tensorflow/models/tree/master/research/slim

P.S. I knew there was bug in TOCO (a conversion tool to convert checkpoint to other format), that will prevent us from converting checkpoints from keras to intermediate format. I am not 100% sure if they fix that or not.

closed 2018-11-17 by dmitriykovalev

llealgt, 2019-04-22:
this issue was closed but i never find a solution, do you guys know at this point some way to convert keras model to the aiy vision kit?