google/aiyprojects-raspbian #519: Cannot broadcast messages between Voice Kit and Home Mini speaker

SinAHal, 2018-11-01:
In both the assistant_grpc_demo.py and assistant_library_demo.py examples I tried to broadcast a message from a voice kit to a home mini speaker and nothing happened. 
grpc_demo: the assistant responds with "ok, broadcasting message..." and nothing happens. 
library_demo: assistant responds with "couldn't detect any other speakers on this network"
When I try broadcast a message from the home mini to the aiy kit I get the same response as library_demo ("no other speaker detected")

Looking at the google home app (on iphone) I am able to see my voice kit, though it is displayed as "Not Responding" and device type is "Light".

SinAHal, 2018-11-02:
I was able to change the device type of the voice kit from the "Actions on Google" page. The device type is now set to Speaker, however this has not changed anything and I still cannot broadcast messages between the voice kit and home mini.

SinAHal, 2018-11-03:
The AIY projects image that I'm using is: aiyprojects-2018-08-03.img

Some info about my home mini:
System firmware version: 137090
Cast firmware version: 1.35.137090

dmitriykovalev, 2018-11-04:
That could be a limitation of [Google Assistant SDK](https://developers.google.com/assistant/sdk/overview)  which we use. It's probably better to submit an issue directly to SDK developers: https://github.com/googlesamples/assistant-sdk-python/issues

closed 2018-11-16 by dmitriykovalev