google/aiyprojects-raspbian #525: Upgrading raspbian to aiy-projects-python 1.1 fails

geeksville, 2018-11-17:
Hi,

In another issue I saw mention that new .debs were out with the latest goodness.  So I followed the directions in HACKING.md to add your apt server and did apt-get update/dist-upgrade.

Alas it failed when installing the aiy-projects-python deb.  Here's the relevant output fro apt-get dist-upgrade:

```
Setting up aiy-overlay-voice (1.0-1) ...
Setting up libx11-6:armhf (2:1.6.4-3+deb9u1) ...
Setting up aiy-projects-python (1.1-0) ...
Cloning into '/opt/aiy/projects-python'...
Receiving objects: 100% (2594/2594), 1.53 MiB | 1.86 MiB/s, done.
Resolving deltas: 100% (1195/1195), done.
warning: remote HEAD refers to nonexistent ref, unable to checkout.

/opt/aiy/projects-python /
Note: checking out 'aiy-github'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 5c74ce5... Update HACKING.md.
Deleted tag 'aiy-github' (was 5c74ce5)
Switched to a new branch 'aiyprojects'
/

Obtaining file:///opt/aiy/projects-python/src
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 1025, in check_if_exists
    self.satisfied_by = pkg_resources.get_distribution(str(no_marker))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 553, in get_distribution
    dist = get_provider(dist)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 427, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 854, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (google-cloud-speech 0.30.0 (/usr/local/lib/python3.5/dist-packages), Requirement.parse('google-cloud-speech==0.36.0'), {'aiy-projects-python'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 521, in _prepare_file
    req_to_install.check_if_exists()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 1036, in check_if_exists
    self.req.name
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 553, in get_distribution
    dist = get_provider(dist)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 427, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 854, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (google-cloud-speech 0.30.0 (/usr/local/lib/python3.5/dist-packages), Requirement.parse('google-cloud-speech==0.36.0'), {'aiy-projects-python'})
dpkg: error processing package aiy-projects-python (--configure):
 subprocess installed post-installation script returned error exit status 2
Setting up aiy-bt-prov-server (1.0-1) ...
Created symlink /etc/systemd/system/multi-user.target.wants/aiy-bt-prov-server.service → /lib/systemd/system/aiy-bt-prov-server.service.
Setting up lxpolkit (0.5.3-2+rpi1) ...
Setting up libxcursor1:armhf (1:1.1.14-1+deb9u2) ...
Setting up pcmanfm (1.2.5-3+rpi5) ...
Setting up xsel (1.2.0-2) ...
Setting up libraspberrypi-bin (1.20181112-1) ...
Setting up qt5ct (0.34-1~bpo9+rpt1) ...
Setting up python3-pyperclip (1.5.27-3~deb9u1) ...
Setting up libpam-systemd:armhf (232-25+deb9u6) ...
Setting up lxpanel (0.9.3-1+rpi7) ...
```

I haven't really poked around but I can later if you want extra info.

geeksville, 2018-11-17:
It seems some dependencies are incorrect (or I missed an upgrade step).

I ran the following to fix things then reinstalled from the deb:

```
sudo pip3 install 'google-cloud-speech>=0.36.0'
sudo pip3 install 'google-auth==1.5.1'
sudo pip3 install 'google-assistant-grpc==0.2.0'
sudo dpkg -i aiy-projects-python_1.1-0_all.deb
```

dmitriykovalev, 2018-11-18:
Doing dist-upgrade based on the previous SD card  image is problematic. Ideally you should take our latest SD card image and future upgrades should work (at least for all driver packages). Package `aiy-projects-python` is a different story, I'll post some updates here regarding this package.