google/aiyprojects-raspbian #526: Setting up aiy-python-wheels: protobuf not supported on armv6.1

jannewmarch, 2018-11-18:
I'm trying to setup Mycroft on the Voice Hat with an RPi 3B+. The install script calls 
    sudo apt-get install aiy-python-wheels
This fails with
    protobuf-3.6.1-cp35-cp35m-linux_armv6.1 is not a supported wheel on this platform
Is this an AIY issue? Is there a fix or workaround?

Thanks, Jan

dmitriykovalev, 2018-11-18:
Thanks for reporting this. Technically this wheel is compatible and should work on RPi 3B+ but looks like `pip` is not happy about its name. Please use [our SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116) if possible, it should work on any Pi board.

xlyytcy, 2018-11-20:
Same issue.

genus1, 2018-11-21:
This is a challenge.  I had five boards working before the upgrade to button.wait_for_press().  I have a LOT of stuff on the pi besides just voice to be able to control boards with voice.  I literally have to start from scratch to rebuild these boards, which is really not a desired option.  Please post an upgrade here when the bug is fixed using upgrade.  Or in the meantime could you post the code to fix the bug you put in your new image?

jannewmarch, 2018-11-22:
andlo has posted a fix in the Mycroft forum [Setting up aiy-python-wheels: protobuf not supported on armv6.1 ](https://community.mycroft.ai/t/setting-up-aiy-python-wheels-protobuf-not-supported-on-armv6-1/5130/4)

genus1, 2018-11-22:
This issue should NOT be closed until apt upgrade works properly.  Google needs to fix this or every time we update or upgrade we have to repatch.  Also it is NOT an answer to just download the current image when it overwrites all your code.

novalu, 2018-11-23:
Actually, answer is not to download the current image, because it has the same issue.

I've tried to install packages manually on Raspbian images:
• 2017-09-07
• 2018-03-13
• 2018-11-13
and it didn't worked with the same error as @jannewmarch has.

Then I've gave it up and use flash provided image aiyprojects-latest.img.xz from 2018-08-07, but `apt-get upgrade` didn't work either with the same error. It looks like it will be working without the upgrading (but the I've discovered another issue #356 :/).

I have Raspberry 3 with Voice Kit V1 and everything worked when I've tried installing it for the first time in October 2017. Thank you.

novalu, 2018-11-27:
New image version mentioned manoj7410 in #530 resolved this issue for me.

Cr4z33, 2018-12-06:
How can I apply the new image update without having to reflash it again and loose all my settings?

genus1, 2018-12-20:
I reloaded the new image and it works.  But I used downgrading the RPi.GPIO as mentioned the bottom posts at #481.

mss-deepak, 2019-02-06:
I am using raspberry 3B with AIY voice v1 hat, with this I am getting error on apt install aiy-python-wheels
please suggest me any solution.

jannewmarch, 2019-02-06:
This seems to have been marked closed by me on 22 Nov. If I did, I didn't mean to! I mentioned a workaround, but that isn't a fix. So I'm re-opening it, because it isn't fixed yet.

mss-deepak, 2019-02-07:
> This seems to have been marked closed by me on 22 Nov. If I did, I didn't mean to! I mentioned a workaround, but that isn't a fix. So I'm re-opening it, because it isn't fixed yet.

Thanks for reopening as this issue as is widely faced by users without any working solution.
@jannewmarch do you found any solution for your problem?
I am trying to install this from last 2 weeks, I also tried the recommended solution to update the image with latest one bu this image is not boot able which is another problem :
https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116

mpember, 2019-02-10:
If the problem has been fixed, could you advise what was done to fix it?  Simply telling users to start with a clean install is NOT the most user-friendly approach.

jannewmarch, 2019-02-10:
andlo has posted a workaround for Mycroft in the Mycroft forum [Setting up aiy-python-wheels: protobuf not supported on armv6.1](https://community.mycroft.ai/t/setting-up-aiy-python-wheels-protobuf-not-supported-on-armv6-1/5130/4). It's not a fix - it allows Mycroft to boot up by timing out on aiy-python-wheels, but every hour or so it tries to reload it and has to time out again.

mpember, 2019-02-11:
The problem stems from the aiy-python-wheels package not identifying the hardware prior to forcing an install of an armv61 version of protobuf.

```
#!/bin/bash

set -e

case "$1" in
    configure)
        pip3 install --no-deps --no-cache-dir --disable-pip-version-check \
            /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl

        SITE_PACKAGES=$(python3 -c "import site; print(site.getsitepackages()[0])")

        # armv6l
        SITE_PACKAGES_ARMV6L=${SITE_PACKAGES}-linux-armv6l
        mkdir -p ${SITE_PACKAGES_ARMV6L}
        unzip -d ${SITE_PACKAGES_ARMV6L} \
            /opt/aiy/python-wheels/google_assistant_library-1.0.1-py2.py3-none-linux_armv6l.whl

        # armv7l
        SITE_PACKAGES_ARMV7L=${SITE_PACKAGES}-linux-armv7l
        mkdir -p ${SITE_PACKAGES_ARMV7L}
        unzip -d ${SITE_PACKAGES_ARMV7L} \
            /opt/aiy/python-wheels/google_assistant_library-1.0.1-py2.py3-none-linux_armv7l.whl

        cat > ${SITE_PACKAGES}/aiy.pth <<EOF
import distutils.util; import sys; import site; path='%s-%s' %(site.getsitepackages()[0], distutils.util.get_platform()); {True: [], False: sys.path}.get(path in sys.path).append(path);
EOF
    ;;
    abort-upgrade|abort-remove|abort-deconfigure)
    ;;

    *)
        echo "postinst called with unknown argument \`$1'" >&2
        exit 0
    ;;
esac

exit 0
```

johanhenselmans, 2019-04-24:
Workaround on rapberry-pi3:
first do
sudo ln -s /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv7l.whl

Then in
/var/lib/dpkg/info/aiy-python-wheels.postinst

replace

/opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl

with
/opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv7l.whl

then do:
sudo dpkg --configure -a

rickwookie, 2019-04-24:
> Workaround on rapberry-pi3:
> first do
> sudo ln -s /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv7l.whl
> Then in [...]

I then get:

` $ sudo dpkg --configure -a
Setting up aiy-projects-python (1.1-0) ...
fatal: destination path '/opt/aiy/projects-python' already exists and is not an empty directory.
dpkg: error processing package aiy-projects-python (--configure):
 subprocess installed post-installation script returned error exit status 128
Errors were encountered while processing:
 aiy-projects-python`

johanhenselmans, 2019-04-24:
You are stalled in a previous phase as the `sudo apt-get install aiy-python-wheels` . 

Perhaps 

`sudo dpkg --configure  --force-overwrite --force-overwrite-dir -a`

works for you? 

Another solution might be to look at the /opt/aiy/ folder, and look at the aiy packages that have already been installed, remove them and start all over again. 

In the end I had these installed:

> ii  aiy-board-info                        1.0-1                             all          AIY Kit Information
ii  aiy-bt-prov-server                    1.0-1                             all          AIY Bluetooth provisioning server
ii  aiy-cwc-server                        1.0-1                             all          AIY Coding with Chrome Server
ii  aiy-dkms                              1.1-2                             all          Hardware device drivers for AIY Projects Kits.
ii  aiy-io-mcu-firmware                   1.0-1                             all          AIY MCU Firmware
ii  aiy-models                            1.1-0                             all          Binary model data for AIY devices
ii  aiy-overlay-vision                    1.0-1                             all          Device tree overlay for AIY Vision Bonnet.
ii  aiy-overlay-voice                     1.0-1                             all          Device tree overlay for AIY Voice Bonnet.
ii  aiy-projects-python                   1.1-0                             all          AIY Python API
ii  aiy-python-wheels                     1.1-1                             all          Prebuilt Python wheels for AIY
ii  aiy-usb-gadget                        1.0-1                             all          AIY USB Gadget
ii  aiy-vision-dkms                       1.1-1                             all          AIY Vision DKMS driver.
ii  aiy-vision-firmware                   1.2-0                             all          Binary firmware for AIY vision bonnet
ii  aiy-voice-services                    1.1-1                             all          AIY Voice Services
ii  aiy-voicebonnet-soundcard-dkms        1.0-2                             all          AIY Voice DKMS driver.

rickwookie, 2019-04-24:
> You are stalled in a previous phase as the sudo apt-get install aiy-python-wheels .
> Perhaps
> sudo dpkg --configure  --force-overwrite --force-overwrite-dir -a
> works for you?
> Another solution might be to look at the /opt/aiy/ folder, and look at the aiy packages that have already been installed, remove them and remove /opt/aiy and start all over again.
> In the end I had these installed:
> 
> ii  aiy-board-info                        1.0-1                             all          AIY Kit Information
> ii  aiy-bt-prov-server                    1.0-1                             all          AIY Bluetooth provisioning server
> ii  aiy-cwc-server                        1.0-1                             all          AIY Coding with Chrome Server
> ii  aiy-dkms                              1.1-2                             all          Hardware device drivers for AIY Projects Kits.
> ii  aiy-io-mcu-firmware                   1.0-1                             all          AIY MCU Firmware
> ii  aiy-models                            1.1-0                             all          Binary model data for AIY devices
> ii  aiy-overlay-vision                    1.0-1                             all          Device tree overlay for AIY Vision Bonnet.
> ii  aiy-overlay-voice                     1.0-1                             all          Device tree overlay for AIY Voice Bonnet.
> ii  aiy-projects-python                   1.1-0                             all          AIY Python API
> ii  aiy-python-wheels                     1.1-1                             all          Prebuilt Python wheels for AIY
> ii  aiy-usb-gadget                        1.0-1                             all          AIY USB Gadget
> ii  aiy-vision-dkms                       1.1-1                             all          AIY Vision DKMS driver.
> ii  aiy-vision-firmware                   1.2-0                             all          Binary firmware for AIY vision bonnet
> ii  aiy-voice-services                    1.1-1                             all          AIY Voice Services
> ii  aiy-voicebonnet-soundcard-dkms        1.0-2                             all          AIY Voice DKMS driver.

I renamed the projects-python directory to projects-python2 then re-ran:
`$ sudo dpkg --configure -a
Setting up aiy-projects-python (1.1-0) ...
Cloning into '/opt/aiy/projects-python'...
Receiving objects: 100% (2594/2594), 1.53 MiB | 0 bytes/s, done.
Resolving deltas: 100% (1195/1195), done.
warning: remote HEAD refers to nonexistent ref, unable to checkout.

/opt/aiy/projects-python /
Note: checking out 'aiy-github'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 5c74ce5... Update HACKING.md.
Deleted tag 'aiy-github' (was 5c74ce5)
Switched to a new branch 'aiyprojects'
/
Obtaining file:///opt/aiy/projects-python/src
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 1025, in check_if_exists
    self.satisfied_by = pkg_resources.get_distribution(str(no_marker))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 553, in get_distribution
    dist = get_provider(dist)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 427, in get_provider
    return working_set.find(moduleOrReq) or require(str(moduleOrReq))[0]
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 854, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (google-cloud-speech 0.30.0 (/usr/local/lib/python3.5/dist-packages), Requirement.parse('google-cloud-speech==0.36.0'), {'aiy-projects-python'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 521, in _prepare_file
    req_to_install.check_if_exists()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 1036, in check_if_exists
    self.req.name
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 553, in get_distribution
    dist = get_provider(dist)
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 427, in get_provider
[... cut ...]
  File "/usr/share/python-wheels/pkg_resources-0.0.0-py2.py3-none-any.whl/pkg_resources/__init__.py", line 854, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (google-cloud-speech 0.30.0 (/usr/local/lib/python3.5/dist-packages), Requirement.parse('google-cloud-speech==0.36.0'), {'aiy-projects-python'})
dpkg: error processing package aiy-projects-python (--configure):
 subprocess installed post-installation script returned error exit status 2
Errors were encountered while processing:
 aiy-projects-python
`

Whytehorse, 2019-05-10:
Solved this. Someone hard-coded an older version of protobuf into /var/lib/dpkg/info/aiy-python-wheels.postinst . Please send this upstream for correction. The workaround is as follows:

pip3 download protobuf
(file downloaded is protobuf-3.7.1-py2.py3-none-any.whl)
 pip3 install ./protobuf-3.7.1-py2.py3-none-any.whl

sudo nano /var/lib/dpkg/info/aiy-python-wheels.postinst

Comment out the lines containing 
# pip3 install --no-deps --no-cache-dir --disable-pip-version-check \
       #     /opt/aiy/python-wheels/protobuf-3.6.1-cp35-cp35m-linux_armv6l.whl

Finish installation:
sudo dpkg --configure --force-overwrite --force-overwrite-dir -a

MaximeSoulie, 2019-08-13:
> Solved this. Someone hard-coded an older version of protobuf into /var/lib/dpkg/info/aiy-python-wheels.postinst . Please send this upstream for correction. The workaround is as follows:
>
> ```
>
> pip3 [...]

Thank you very much Whytehorse, as this issue is still present on the latest image 2019-07-20... !

Avi-avidan, 2019-11-11:
trying to install aiy v1 [python-wheels (1.1-1)] on rpi4. followed this entire thread. workarounds did not work for me :-/.
any advice would be greatly appreciated. really.

sudo dpkg --configure --force-overwrite --force-overwrite-dir -a
`Setting up aiy-voicebonnet-soundcard-dkms (1.0-2) ...
Removing old aiy-voicebonnet-soundcard-1.0 DKMS files...
Deleting module version: 1.0
completely from the DKMS tree.
Done.
Loading new aiy-voicebonnet-soundcard-1.0 DKMS files...
It is likely that 4.19.75-v7l+ belongs to a chroot's host
Building for 4.19.75+, 4.19.75-v7+, 4.19.75-v7l+ and 4.19.75-v8+
Building initial module for 4.19.75+
Error! Bad return status for module build on kernel: 4.19.75+ (armv7l)
Consult /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/make.log for more information.
dpkg: error processing package aiy-voicebonnet-soundcard-dkms (--configure):
 installed aiy-voicebonnet-soundcard-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 aiy-voicebonnet-soundcard-dkms`

cat /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/make.log
`DKMS make.log for aiy-voicebonnet-soundcard-1.0 for kernel 4.19.75+ (armv7l)
Mon 11 Nov 2019 10:56:18 PM IST
make: Entering directory '/usr/src/linux-headers-4.19.75+'
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rl6231.o
In file included from /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.c:25:
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.h:2207:36: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
 int rt5645_sel_asrc_clk_src(struct snd_soc_codec *codec,
                                    ^~~~~~~~~~~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.h:2210:35: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
 int rt5645_set_jack_detect(struct snd_soc_codec *codec,
                                   ^~~~~~~~~~~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_init’:
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.c:40:29: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec’
  rt5645_sel_asrc_clk_src(rtd->codec,
                             ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.c:62:35: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec’
  return rt5645_set_jack_detect(rtd->codec, &headset_jack, NULL, NULL);
                                   ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.c:63:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[1]: *** [scripts/Makefile.build:309: /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/snd-aiy-voicebonnet.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:35:
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.h:2207:36: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
 int rt5645_sel_asrc_clk_src(struct snd_soc_codec *codec,
                                    ^~~~~~~~~~~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.h:2210:35: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
 int rt5645_set_jack_detect(struct snd_soc_codec *codec,
                                   ^~~~~~~~~~~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c: In function ‘rt5645_reset’:
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:428:9: error: implicit declaration of function ‘snd_soc_write’; did you mean ‘snd_ac97_write’? [-Werror=implicit-function-declaration]
  return snd_soc_write(codec, RT5645_RESET, 0);
         ^~~~~~~~~~~~~
         snd_ac97_write
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c: In function ‘set_dmic_clk’:
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:835:32: error: implicit declaration of function ‘snd_soc_dapm_to_codec’; did you mean ‘snd_soc_dapm_to_component’? [-Werror=implicit-function-declaration]
  struct snd_soc_codec *codec = snd_soc_dapm_to_codec(w->dapm);
                                ^~~~~~~~~~~~~~~~~~~~~
                                snd_soc_dapm_to_component
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:835:32: warning: initialization of ‘struct snd_soc_codec *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:836:31: error: implicit declaration of function ‘snd_soc_codec_get_drvdata’; did you mean ‘snd_soc_dai_get_drvdata’? [-Werror=implicit-function-declaration]
  struct rt5645_priv *rt5645 = snd_soc_codec_get_drvdata(codec);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~
                               snd_soc_dai_get_drvdata
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:836:31: warning: initialization of ‘struct rt5645_priv *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
In file included from ./include/linux/i2c.h:30,
[... 241 lines cut ...]
[... cut ...]
/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.c:3499:36: error: storage size of ‘soc_codec_dev_rt5645’ isn’t known
 static struct snd_soc_codec_driver soc_codec_dev_rt5645 = {
                                    ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:309: /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/rt5645.o] Error 1
make: *** [Makefile:1519: _module_/var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.75+'`

dmitriykovalev, 2019-11-18:
This should be fixed in the latest [SD card image](https://github.com/google/aiyprojects-raspbian/releases), please follow [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) to install `python-wheels [1.3-1]`.

closed 2019-11-18 by dmitriykovalev

d8ahazard, 2020-01-04:
> This should be fixed in the latest [SD card image](https://github.com/google/aiyprojects-raspbian/releases), please follow [HACKING.md](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) to install `python-wheels [1.3-1]`.

Yeah, that's a big negatory there, ghost rider. Getting the same issue when I try to install manually:

pi@raspberrypi:~ $ sudo apt-get install -y aiy-python-wheels
Reading package lists... Done
Building dependency tree       
Reading state information... Done
aiy-python-wheels is already the newest version (1.3-1).
The following packages were automatically installed and are no longer required:
  vlc-plugin-notify vlc-plugin-samba vlc-plugin-video-splitter vlc-plugin-visualization
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 17 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up aiy-python-wheels (1.3-1) ...
protobuf-3.10.0-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
dpkg: error processing package aiy-python-wheels (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 aiy-python-wheels
E: Sub-process /usr/bin/dpkg returned an error code (1)

combs, 2020-02-12:
```
Setting up aiy-python-wheels (1.3-1) ...
protobuf-3.10.0-cp37-cp37m-linux_armv6l.whl is not a supported wheel on this platform.
```