google/aiyprojects-raspbian #527: Running `apt-get upgrade` prevents the AIY demos from functioning

wildtangent, 2018-11-20:
Using the Vision kit + Pi Zero, after installing the latest AIY Raspian release ISO, and running `apt-get update`, either via command line or via the installer prompt during first boot, after the update are all finished, the joy detector demo fails to launch.

There are a number of packages pertaining to the AIY which are uninstalled during the apt-get upgrade process, such as the leds, aiy etc

When trying to run the joy detector, the process hangs for some seconds then aborts.
In previous attempts I have also had notification that the program cannot access the LED

I've re-flashed the ISO again so I don't have access to the precise error messages at present but can recreate if necessary

choconnit, 2018-12-02:
I have the same issue running the latest 2018_11_16 image on Vision Kit v1.1.  
Joy Detector Demo failed on an instantiation of the Leds class, looking for the non-existent DEVICE_PATH = '/sys/class/leds/ktd202x:led1/device'

I'm stuck.  What does re-flashed the ISO entailed?

manoj7410, 2018-12-02:
@choconnit You can download the sdcard image from : https://github.com/google/aiyprojects-raspbian/releases/tag/v20181116 and install it on your sdcard by using Etcher : https://www.balena.io/etcher/
apt-get upgrade might have some issue which is breaking the AIY modules.

perronj2, 2018-12-05:
@manoj7410 I had a the same problem after apt-get update & apt-get upgrade. 

After reinstalling the latest AIY image (11/16/18) and NOT doing and update I was able to get the AIY modules to run.

asmagill, 2018-12-06:
This has come up a number of times in various issues... has anyone tracked it to specific packages which can be held back rather than just saying "don't run apt upgrade"?

seth-johnson-sp, 2018-12-08:
> @manoj7410 I had a the same problem after apt-get update & apt-get upgrade.
> 
> After reinstalling the latest AIY image (11/16/18) and NOT doing and update I was able to get the AIY modules to run.

I concur with this experience. Not-updating the Raspbian OS is the only way to ensure the AIY stuff will launch.

branflake2267, 2019-01-01:
I created a video on flashing the micro sd card it works great if you want to flash to the latest image available on github. I fixed it by flashing to 11.16 at the time of this post. 
https://www.youtube.com/watch?v=WV4X_8OLJF8

ayman, 2019-01-12:
@choconnit 
> Joy Detector Demo failed on an instantiation of the Leds class, looking for the non-existent DEVICE_PATH = '/sys/class/leds/ktd202x:led1/device'

The fix for LEDS and PWM is detailed in #385 specifically: https://github.com/google/aiyprojects-raspbian/issues/385#issuecomment-392231547

Which, was closed but the error still is there on the lasted build (11/2018).

atmape, 2019-01-19:
I bought AIY Voice Kit a few month back and I tried it out the first time today and ran into multiple Problems. With original Image running at first I didn't got the demo running, but I found out it was a hardware issue because the cables of the speaker got too much bend in box, so I had to shorten the cables a little bit (but that I found out in the end).

At first I tried `aplay /usr/share/sounds/alsa/Front_Center.wav` to see if the speaker worked. It worked. Then I tried if the microphone work  with `arecord -d 5 test.wav`. The recording work, but after recording I wasn't able to play it with `aplay` because it crashed. I thought maybe updating will help. But after updating and rebooting I wasn't able to play the testsound with  `aplay /usr/share/sounds/alsa/Front_Center.wav`.

Usally soundcard will be found:
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: aiyvoicebonnet [aiy-voicebonnet], device 0: Google AIY Voice Bonnet SoundCard HiFi rt5645-aif1-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: aiyvoicebonnet [aiy-voicebonnet], device 0: Google AIY Voice Bonnet SoundCard HiFi rt5645-aif1-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

But after upgrading you only see:
```
pi@raspberrypi:~ $ aplay -l
aplay: device_list:270: no soundcards found...
```

The Problem is that during upgrade a newer kernel is installed, but the aiy specific kernel modules aren't (re)build during the upgrade process. So after reboot the kernel is able to load them. The deb packages are installed so the only thing you have to do is rebuild them:

```
sudo dpkg-reconfigure aiy-board-info aiy-bt-prov-server aiy-cwc-server aiy-dkms aiy-io-mcu-firmware aiy-models aiy-overlay-vision aiy-overlay-voice aiy-projects-python aiy-python-wheels aiy-usb-gadget aiy-vision-dkms aiy-vision-firmware aiy-voice-services aiy-voicebonnet-soundcard-dkms leds-ktd202x-dkms pwm-soft-dkms
```

I'm not sure if you really need all of them but I `grep` all `aiy` from `dpkg -l`. After about 30min the modules are rebuild und you can `sudo shutdown -r now`. After that I had an updated raspberry with a functioning Voice KIT.

rickyklu, 2019-05-28:
Similarly I ran `sudo apt-get update && sudo apt-get upgrade` and now my RPi Voice V1 Kit  speakers or microphone no longer work.  I am running having the same `aplay` errors that @atmape had.
So I tried running the long `sudo dpkg-reconfigure ...` command. But ran into some errors. I had to ommit a couple of packages because I didn't need all of them like python-wheels, but it still runs into an error when it tries to reconfigure the `aiy-voicebonnet-soundcard-dkms leds-ktd202x-dkms pwm-soft-dkms` packages. 
(My RPi's current kernel is 4.19.42-v7+)

```
pi@raspberrypi:~ $ sudo dpkg-reconfigure aiy-voicebonnet-soundcard-dkms
------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.
Loading new aiy-voicebonnet-soundcard-1.0 DKMS files...
It is likely that 4.19.42-v7+ belongs to a chroot's host
Building for 4.19.42+ and 4.19.42-v7+
Building initial module for 4.19.42+
Error! Bad return status for module build on kernel: 4.19.42+ (armv7l)
Consult /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/make.log for more information.
```
[Result of /var/lib/dkms/aiy-voicebonnet-soundcard/1.0/build/make.log](https://pastebin.com/0jLzWALU)
My guess is that the voicehat doesn't work well with my current kernel. Is there any way to make the aiy-hat play nice with the current kernel?