google/aiyprojects-raspbian #529: Getting Pin number or Bonnet Pin Access for Node.js

ankurp, 2018-11-21:
How can we read from the GPIO pins via node.js

I want to connect a PIR motion sensor to this and have the power and ground wires connect and connected to Pin A but in nodes library I need to set the mode and the PIN number. What is the mode and the numeric PIN number?

ankurp, 2018-12-05:
Any update. I am looking to connect to the bonnet GPIO pin but cannot use Python as my library for it. I want to connect a PIR motion sensor.

dmitriykovalev, 2018-12-06:
You can use `sysfs` in order to control GPIO pins. There are many documentation sources, e.g. https://elinux.org/GPIO. To export our 4 pins you need to run:

```bash
pi@raspberrypi:~ $ echo $((497 + 2)) > /sys/class/gpio/export
pi@raspberrypi:~ $ echo $((497 + 3)) > /sys/class/gpio/export
pi@raspberrypi:~ $ echo $((497 + 8)) > /sys/class/gpio/export
pi@raspberrypi:~ $ echo $((497 + 9)) > /sys/class/gpio/export
```

You'll see exported pins as:

```
/sys/class/gpio/AIY_USER0
/sys/class/gpio/AIY_USER1
/sys/class/gpio/AIY_USER2
/sys/class/gpio/AIY_USER3
```

closed 2018-12-19 by dmitriykovalev