google/aiyprojects-raspbian #532: AIY USB Gadget deletes ETH0

swhobbit, 2018-11-25:
System: 2018-11-16 AIY image on mid 2018 Raspberry Pi Zero W with a SMAYS Ethernet USB Ethernet adapter directly plugged directly into unit.

Upon boot, DHCP server shows both `ETH0` and `WLAN0` requested IPv4 addresses, but `ETH0` cannot be used by SSH or other services.  Inspection via `ifconfig` shows only devices LO and USB -- no ETH0.   ETH0 has been deleted, with no log messages in `syslog`, `daemon.log` or `dmesg` reporting why. 

** Workaround **
Disabling the AIY Gadget service and rebooting machine caused ETH0 to reappear.

***expected behavior***
  * Use of ETH0 is honored when such as device is attached without disabling services.  That is, the AIY gadget service must **not** hijack the USB port and/or disable Ethernet or other useful devices which are directly attached.
  * The AIY gadget reports all actions it takes to the system log.  For example, it must not take actions against the USB port without logging.

swhobbit, 2018-11-25:
Also:
The default setup with  USSB/dnsmasq screws up DNS setup via DHCP, in particular it makes the DNS lookups ignore the DHCP assigned name servers and whatever local information they contain. 

The `dnsmasq` daemon should be **not** activated if USB is not used for networking and `ETH0` **is**.

dmitriykovalev, 2018-11-30:
@swhobbit I'll try to reproduce that. We don't do anything specific to remove `eth0`.